Bepacom-Raalte/bepacom-HA-Addons #11: Cannot write

mikihacia, 2023-05-09:
Hi,

While trying to get the Binary Inputs being detected by the integration, I've found that I cannot write to any BACNet object out there. Commands looking to be sent out, but no value change can be observed at the controller. Does this have anything to deal with the priority? See logs below:

DEBUG:      Received: {'device:2098177': {'analogValue:14': {'presentValue': 18.8}}} in write_property()
INFO:     192.168.0.26:0 - "POST /apiv1/device%3A2098177/analogValue%3A14?deviceid=device:2098177&objectid=analogValue:14&presentValue=18.8 HTTP/1.0" 200 OK
DEBUG:    > TEXT '{"device:1050574": {}, "device:2098177": {"devi..."numberOfStates": 24}}}' [15500 bytes]
DEBUG:      Received: {'device:2098177': {'analogValue:14': {'presentValue': 18.9}}} in write_property()
INFO:     192.168.0.26:0 - "POST /apiv1/device%3A2098177/analogValue%3A14?deviceid=device:2098177&objectid=analogValue:14&presentValue=18.9 HTTP/1.0" 200 OK
DEBUG:      Received: {'device:2098177': {'analogValue:14': {'presentValue': 19.0}}} in write_property()
INFO:     192.168.0.26:0 - "POST /apiv1/device%3A2098177/analogValue%3A14?deviceid=device:2098177&objectid=analogValue:14&presentValue=19.0 HTTP/1.0" 200 OK
INFO:      Writing 18.8 to ('analogValue', 14) presentValue

Thanks!

mikihacia, 2023-05-09:
Is that priority written in the code somewhere, so I could change it to something higher, i.e. Priority 8 (Manual Operation)?
I could get away with using just one, but higher priority for my points.

Bepacom-Raalte, 2023-05-09:
It can be added yes! 
If you go to /rootfs/usr/bin/BACnetIOHandler.py and look for the WriteProperty() function, you'll see:

`
    request = WritePropertyRequest(
        objectIdentifier=objectID,
        propertyIdentifier=propertyID,
        propertyArrayIndex=None,
        propertyValue=None,
        priority=None,
        )
`

If you adjust priority to any integer you want, every message you try to send should have that priority.
If you download the add-on repository and install it as a local add-on, you'll be able to try more things. 
All BACnet things are happening in this file, so it's most important for proper functioning of the add-on.

mikihacia, 2023-05-09:
That sounds promising, but I don't have a clue how to get to the addon's file system.

What do you mean by installing the add-on locally? Just copy the repo to /addons folder and restart HA?

I've never got that deep into this?

Bepacom-Raalte, 2023-05-09:
Yes, it's basically that. Through SMPT your /addons folder is accessible to you. There you can drop the entire repository in, or just the bacnetinterface folder. When you refresh the add-on store it should appear in the store as a local add-on. Any changes you make before you install will get installed in the local version. If you adjust anything after that, you'll have to reinstall the add-on before the changes have any effect.
Be sure to shut down the add-on downloaded from the repository, or they'll prevent starting the local add-on. Hope this helps, otherwise I'll make an option to set the priority from the web UI.

mikihacia, 2023-05-09:
Just works like a charm!

If I may have a suggestion: add the priority option as the entity attribute, maybe? I'm not quite sure how it exactly works, but the default one could be 15, and there will be always an option to increase it. Alongside overriding the priority it might be worth considering on how to release them as well, maybe a service or through WebUI.

Other suggestion may be using the object description as a automatic Friendly Name generation, apart from the name being generated from the objectName. This might be a request for the integration though.

And last, but not least: would be great to have an option to decide if the point should be either switch or binary input in the HA. This is because some of the BVAL points are presenting just the state, so there's no need to switch them accidentally. This one might be quite hard to do I presume, but I thought I'll leave it here.

Just wanna to say big thanks for the huge piece of the software you've guys made and for the prompt and accurate support on the issues! THANKS!

Bepacom-Raalte, 2023-05-09:
Wonderful!

I'll take your suggestions in consideration, I do think the priority should be adjustable from at least the web UI. Through Home Assistant entities it'll be difficult a little as there's only a few things that can be adjusted in entities.

I understand the issue with Binary Values, it's similar to Analog Value objects as some of those just display a number as well. Treating them as switch and number means they'll always be able to be displayed and written to whenever possible. Perhaps in the future if it's a little easier to allow choosing entity types in Home Assistant I can integrate that into the integration.

The problem with friendly name generation is that the description is not always telling for large installations. It was a wish of my company that it'd be generated by the object name instead of descriptions. It is possible to change the names of entities in Home Assistant though, so you can find it by the name you gave it.

Glad I could be of help and thank you for your suggestions! They'll definitely be of help to us. I'll close this issue when I added some way to add priority to writing.

GravySeal, 2023-05-09:
I added a config option where you can set the priority for the write request. Default is 15. In the future I want to make it dynamic, but for now this at least enables changing the priority of messages!

closed 2023-05-09 by GravySeal