Bepacom-Raalte/bepacom-HA-Addons #24: [Future Request] Create BACNet entities

mikihacia, 2023-10-09:
I was just wondering if it would be possible to add a feature to create entities that'll be available through BACNet or assign the existing ones to the Export Table?

Unless, this is already available? 🤔

Bepacom-Raalte, 2023-10-31:
Hello,

I just came back from holidays, so apologies for replying late. I've been thinking about this feature as well.
Currently we don't have much of an use case for this, but perhaps when I have some spare time I could implement it.

It could be that you can write down what entities you want in the settings of the add-on. This would make the BACnet device of the add-on create objects that represent these entities in the BACnet network.

Let me know how you think it should be implemented!

mikihacia, 2023-10-31:
> It could be that you can write down what entities you want in the settings of the add-on. This would make the BACnet device of the add-on create objects that represent these entities in the BACnet network.

This is exactly what I've meant. 

It can be used with various types of devices (especially PLC) and would save hass from processing these in automations.

I was also wondering if there's any way to determinate when the device has lost comms (like disconnected cable of loss of power), unless it's already in.

Thanks!

Bepacom-Raalte, 2023-10-31:
> I was also wondering if there's any way to determinate when the device has lost comms (like disconnected cable of loss of power), unless it's already in.

Currently I don't really keep track of it. I assume the device is there, and poll every minute. Perhaps in the future I can make it a feature to make some kind of alert for it.

ztardik, 2023-12-15:
> 
> Currently I don't really keep track of it. I assume the device is there, and poll every minute. Perhaps in the future I can make it a feature to make some kind of alert for it.

Could you use COV-subscriptions instead of polling? Then poll every 15-30 minutes for a periodic update.
And to check for a device just send a who-is periodically. If no response then declare all points from that device invalid.

Bepacom-Raalte, 2023-12-15:
> Could you use COV-subscriptions instead of polling? Then poll every 15-30 minutes for a periodic update.

This is what's currently happening, a CoV subscription as the main way to get data and a periodic polling as a backup if the CoV fails. The polling rate is configurable, but by default 60 seconds.

> And to check for a device just send a who-is periodically. If no response then declare all points from that device invalid.

This is a pretty good idea, might implement this.

Bepacom-Raalte, 2023-12-18:
> If no response then declare all points from that device invalid.

I've been thinking about sending a periodic Who Is request. Once one device fails to respond, what course of action would be appropriate? 
It could be that the API just removes that device from the list, and adds it back once it gets a new I Am request.
Another way is to just not update the API points, but keep them available for polling.

GravySeal, 2024-01-02:
> I was just wondering if it would be possible to add a feature to create entities that'll be available through BACNet

Just made a start with this, it's out in the dev version if you'd like to play around with it. Not possible to write yet though.

mikihacia, 2024-01-02:
> > I was just wondering if it would be possible to add a feature to create entities that'll be available through BACNet
> 
> @mikihacia I just made a start with this, it's out in the dev version if you'd like to play around with it. Not possible to write yet though.

Just had a chance to play with it and it's been working great so far. Looking forward to test the write function when available. 

Doing brilliant job, guys! This stuff should be a part of the core of HASS as the modbus is now.

GravySeal, 2024-01-07:
Writing to present value and out of service should work now, as well as immediate updates when values change. Configuration changed a little bit, so once you reinstall or rebuild after checking updates please check.

mikihacia, 2024-01-11:
I briefly checked and it works like a charm!

This is amazing 🤩.  Thanks!

Bepacom-Raalte, 2024-04-03:
I'll close this issue, for feature requests Discussions is a great place!