Bepacom-Raalte/bepacom-HA-Addons #28: Unable to connect to device

magnusoverli, 2023-12-02:
Hi!

I have installed the addon, and manually set the IP-address of the network interface. 

I get no errors in the addon log (see bottom).
I also get no errors in the Supervisor log.

From wireshark I can see that the addon broadcasts a "i-Am device, 2" message and a subsequent "who-Is" message. This happens every time I restart the addon (as expected) and the device I want to communicate with responds with a "i-Am-device, 3" message.

The device is still not showing up in the addon GUI...?

What else can I troubleshoot?

The device I want to communicate with is a Flexit Nordic CI3 ventilation unit.

**WIRESHARK OUTPUT**
114796	723.375441	192.168.5.33	192.168.5.255	BACnet-APDU	66	Unconfirmed-REQ i-Am device,3 
114797	723.375441	192.168.5.33	192.168.5.255	BACnet-APDU	60	Unconfirmed-REQ who-Is 
114809	723.672403	192.168.5.25	255.255.255.255	BACnet-APDU	62	Unconfirmed-REQ i-Am device,2 
231819	1374.025193	192.168.5.33	192.168.5.255	BACnet-APDU	66	Unconfirmed-REQ i-Am device,3 
231820	1374.025193	192.168.5.33	192.168.5.255	BACnet-APDU	60	Unconfirmed-REQ who-Is 
231821	1374.025193	192.168.5.25	255.255.255.255	BACnet-APDU	62	Unconfirmed-REQ i-Am device,2 

**ADDON LOG**
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Bepacom EcoPanel BACnet/IP Interface
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.1.3
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2.rc2  (amd64 / qemux86-64)
 Home Assistant Core: 2023.12.0b1
 Home Assistant Supervisor: 2023.11.6
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[19:42:20] INFO: Initialising NGINX
[19:42:20] INFO: Generating BACpypes.ini
s6-rc: info: service init-nginx successfully started
Using enp1s0 as address: 192.168.5.33/24
[BACpypes]
objectName: HA
address: 192.168.5.33/24
objectIdentifier: 3
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: WARNING
defaultPriority: 1
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
[19:42:20] INFO: Running interface
s6-rc: info: service legacy-services successfully started
WARNING:  Invalid HTTP request received.
[19:42:26] INFO: Starting NGINX
2023/12/02 19:42:26 [notice] 270#270: using the "epoll" event method
2023/12/02 19:42:26 [notice] 270#270: nginx/1.24.0
2023/12/02 19:42:26 [notice] 270#270: OS: Linux 6.1.63-haos
2023/12/02 19:42:26 [notice] 270#270: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/02 19:42:26 [notice] 270#270: start worker processes
2023/12/02 19:42:26 [notice] 270#270: start worker process 319

magnusoverli, 2023-12-02:
Hey!

Sure, here it is:

The device IP is 192.168.5.25, as seen in the wireshark log.
Here is an INFO-level addon log:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------

 Add-on: Bepacom EcoPanel BACnet/IP Interface
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.1.3
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2.rc2  (amd64 / qemux86-64)
 Home Assistant Core: 2023.12.0b1
 Home Assistant Supervisor: 2023.11.6
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[23:13:10] INFO: Initialising NGINX
[23:13:10] INFO: Generating BACpypes.ini
s6-rc: info: service init-nginx successfully started
Using enp1s0 as address: 192.168.5.33/24
[BACpypes]
objectName: HA
address: 192.168.5.33/24
objectIdentifier: 3
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: INFO
defaultPriority: 1
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
[23:13:10] INFO: Running interface
s6-rc: info: service legacy-services successfully started
INFO:     Started server process [266]
INFO:     Waiting for application startup.
INFO:    Refreshing subscriptions...
INFO:     Application startup complete.
INFO:     Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
WARNING:  Invalid HTTP request received.
[23:13:16] INFO: Starting NGINX
2023/12/02 23:13:16 [notice] 271#271: using the "epoll" event method
2023/12/02 23:13:16 [notice] 271#271: nginx/1.24.0
2023/12/02 23:13:16 [notice] 271#271: OS: Linux 6.1.63-haos
2023/12/02 23:13:16 [notice] 271#271: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/02 23:13:16 [notice] 271#271: start worker processes
2023/12/02 23:13:16 [notice] 271#271: start worker process 320
```

GravySeal, 2023-12-03:
I know there can be some issues when using the NGINX add-on, but yours seem to be fine.
Could you try to use 192.168.5.33/0 as your IP in the configuration to see what happens?

magnusoverli, 2023-12-03:
Hey ho! That made it work! 👍 

Would you care to explain what the difference is and why it was not working before?

magnusoverli, 2023-12-03:
Thanks for your assistance and explanation! Much appreciated!

Related question:
In the accompanying integration, is it possible to have a config option to set wether to use the objectIdentifier or the object description as the name of the entity? "R(1)'TSu" does not make much sense but "Supply Air Temperature" does... :-)

magnusoverli, 2023-12-03:
That worked great! Much easier to sort through the thousands (!) of datapoints when they have a descriptive name...

Consider making that a configurable option for those less tech inclined? ;-)

Thanks for all the help. Will close the issue now. 👍