Bepacom-Raalte/bepacom-HA-Addons #30: exception: KeyError - Alerton VLCA-1688

azzey1, 2023-12-17:
**Describe the bug**
Attempting to connect an Alerton VLCA-1688 to HA Bepacom EcoPanel BACnet/IP Interface Development Version V1.2.0
Detects device in log, No device shown in side bar interface.

Alerton Field device instance: 2000, IP 192.168.1.211/24

HA Details:
Core 2023.12.3
Supervisor 2023.11.6
Operating System 11.2
Frontend 20231208.2

Log file 
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------

 Add-on: Bepacom EcoPanel BACnet/IP Interface Development Version
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.2.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2023.12.3
 Home Assistant Supervisor: 2023.11.6
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[18:19:50] INFO: Running init-nginx.
[18:19:50] INFO: Running init-interface.
[18:19:50] INFO: Allowed interfaces for NGINX: 127.0.0.1/8 192.168.1.22/24 172.30.32.1/23 172.30.232.1/23
[18:19:50] INFO: Generated ingress configuration!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.1.22/24
objectIdentifier: 420
maxApduLengthAccepted: 1476
segmentation: noSegmentation
vendorIdentifier: 1234
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: DEBUG
defaultPriority: 8
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
[18:19:51] INFO: Running interface
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
DEBUG:    Application initialised
INFO:     Started server process [271]
INFO:     Waiting for application startup.
INFO:    I Am from device,2000
DEBUG:    Exploring Device info of device,2000
ERROR:    Abort PDU error while reading device properties: device,2000: segmentation-not-supported
INFO:    Reading objectList from device,2000...
ERROR:    exception: KeyError('device:2000')
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/app.py", line 899, in indication
    await helperFn(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 139, in do_IAmRequest
    await self.read_object_list(device_identifier=apdu.iAmDeviceIdentifier)
  File "/usr/bin/BACnetIOHandler.py", line 256, in read_object_list
    for obj_id in self.bacnet_device_dict[f"device:{device_identifier[1]}"][
                  ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'device:2000'
ERROR:    An error occurred: 'IAmRequest' object has no attribute 'apduInvokeID'
INFO:     Application startup complete.
INFO:     Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
WARNING:  Invalid HTTP request received.
[18:19:58] INFO: Running NGINX
2023/12/17 18:19:58 [notice] 281#281: using the "epoll" event method
2023/12/17 18:19:58 [notice] 281#281: nginx/1.24.0
2023/12/17 18:19:58 [notice] 281#281: OS: Linux 6.1.63-haos
2023/12/17 18:19:58 [notice] 281#281: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/17 18:19:58 [notice] 281#281: start worker processes
2023/12/17 18:19:58 [notice] 281#281: start worker process 326
```

GravySeal, 2023-12-17:
Hey @azzey1,

The problem has to do with your device not supporting segmentation. I made a quick update to bring in support for such devices again. Can you try updating the dev version?

azzey1, 2023-12-18:
I have the Device shown in the web panel, how do i reference the point/outputs? or do i still have an error? 
````
-----------------------------------------------------------
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2023.12.3
 Home Assistant Supervisor: 2023.11.6
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[19:32:08] INFO: Running init-nginx.
[19:32:08] INFO: Running init-interface.
[19:32:08] INFO: Allowed interfaces for NGINX: 127.0.0.1/8 192.168.1.22/24 172.30.232.1/23 172.30.32.1/23
[19:32:08] INFO: Generated ingress configuration!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.1.22/24
objectIdentifier: 420
maxApduLengthAccepted: 1476
segmentation: noSegmentation
vendorIdentifier: 1234
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: DEBUG
defaultPriority: 8
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
[19:32:09] INFO: Running interface
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
DEBUG:    Application initialised
INFO:     Started server process [272]
INFO:     Waiting for application startup.
INFO:    I Am from device,2000
DEBUG:    Exploring Device info of device,2000
WARNING:    Abort PDU error while reading device properties: device,2000: segmentation-not-supported
ERROR:    exception: ValueError('minimum length: 1')
Traceback (most recent call last):
  File "/usr/bin/BACnetIOHandler.py", line 226, in read_device_props
    response = await self.read_property_multiple(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/service/object.py", line 543, in read_property_multiple
    response = await self.request(read_property_multiple_request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
bacpypes3.apdu.AbortPDU: segmentation-not-supported

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/app.py", line 899, in indication
    await helperFn(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 137, in do_IAmRequest
    await self.read_device_props(apdu=apdu)
  File "/usr/bin/BACnetIOHandler.py", line 238, in read_device_props
    await self.read_device_props_non_segmented(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 262, in read_device_props_non_segmented
    response = await self.read_property(
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/service/object.py", line 146, in read_property
    property_value = response.propertyValue.cast_out(property_type)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/constructeddata.py", line 1751, in cast_out
    result = cls.decode(tag_list)
             ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/primitivedata.py", line 1490, in decode
    return cls(value)
           ^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/primitivedata.py", line 628, in __call__
    args = (new_type.cast(args[0]),)  # type: ignore[attr-defined]
            ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/primitivedata.py", line 1392, in cast
    raise ValueError(f"minimum length: {min_length}")
ValueError: minimum length: 1
ERROR:    An error occurred: 'IAmRequest' object has no attribute 'apduInvokeID'
INFO:     Application startup complete.
INFO:     Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
WARNING:  Invalid HTTP request received.
[19:32:15] INFO: Running NGINX
2023/12/18 19:32:15 [notice] 282#282: using the "epoll" event method
2023/12/18 19:32:15 [notice] 282#282: nginx/1.24.0
2023/12/18 19:32:15 [notice] 282#282: OS: Linux 6.1.63-haos
2023/12/18 19:32:15 [notice] 282#282: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/18 19:32:15 [notice] 282#282: start worker processes
2023/12/18 19:32:15 [notice] 282#282: start worker process 327
INFO:    Periodic object reading...
```

Bepacom-Raalte, 2023-12-18:
There's still an error. Could you uninstall, check for updates and then reinstall again? I quickly added some logging that hopefully should give some more information for me.

azzey1, 2023-12-18:
ok, uninstalled, what is the expectation on when the update would be ready?

Bepacom-Raalte, 2023-12-19:
The update is already out. You just have to check for updates and then install again. This update will give me some more information what's going wrong.

azzey1, 2023-12-19:
I have tried to detect updates, and doesn't detect a newer version?

Bepacom-Raalte, 2023-12-19:
Right, the version doesn't get bumped. It's still 1.2.1.

azzey1, 2023-12-20:
ok what tests do you need me to follow?

Bepacom-Raalte, 2023-12-20:
For now, just try to start the add-on again, and paste the logs here. Hopefully I can get some more information from it.
Do you know Yet Another BACnet Explorer? If you do, could you maybe export your device as EDE file for me?

azzey1, 2023-12-20:
ok what log level do you want for the add on? debug?

azzey1, 2023-12-20:
[Alerton Device 2000_EDE.csv](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/13724952/Alerton.Device.2000_EDE.csv)
[Alerton Device 2000_StateTexts.csv](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/13724953/Alerton.Device.2000_StateTexts.csv)

azzey1, 2023-12-20:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------

 Add-on: Bepacom EcoPanel BACnet/IP Interface Development Version
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2023.12.3
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[19:41:37] INFO: Running init-nginx.
[19:41:37] INFO: Running init-interface.
[19:41:37] INFO: Allowed interfaces for NGINX: 127.0.0.1/8 192.168.1.22/24 172.30.232.1/23 172.30.32.1/23
[19:41:37] INFO: Generated ingress configuration!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.1.22/24
objectIdentifier: 1999
maxApduLengthAccepted: 1476
segmentation: noSegmentation
vendorIdentifier: 123
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: DEBUG
defaultPriority: 8
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
[19:41:37] INFO: Running interface
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
DEBUG:    Application initialised
INFO:     Started server process [271]
INFO:     Waiting for application startup.
INFO:    I Am from device,2000
DEBUG:    Exploring Device info of device,2000
WARNING:    Abort PDU error while reading device properties: device,2000: segmentation-not-supported
INFO:    Reading device device,2000, object-identifier
ERROR:    Other error: name 'address' is not defined
INFO:    Reading objectList from device,2000...
ERROR:    exception: KeyError('device:2000')
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/app.py", line 899, in indication
    await helperFn(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 139, in do_IAmRequest
    await self.read_object_list(device_identifier=apdu.iAmDeviceIdentifier)
  File "/usr/bin/BACnetIOHandler.py", line 340, in read_object_list
    for obj_id in self.bacnet_device_dict[f"device:{device_identifier[1]}"][
                  ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'device:2000'
ERROR:    An error occurred: 'IAmRequest' object has no attribute 'apduInvokeID'
INFO:     Application startup complete.
INFO:     Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
WARNING:  Invalid HTTP request received.
[19:41:44] INFO: Running NGINX
2023/12/20 19:41:44 [notice] 281#281: using the "epoll" event method
2023/12/20 19:41:44 [notice] 281#281: nginx/1.24.0
2023/12/20 19:41:44 [notice] 281#281: OS: Linux 6.1.63-haos
2023/12/20 19:41:44 [notice] 281#281: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/20 19:41:44 [notice] 281#281: start worker processes
2023/12/20 19:41:44 [notice] 281#281: start worker process 326
```

azzey1, 2023-12-20:
this was the log from when i conducted a scan/add device with YABE ,
```
KeyError: 'device:2000'
ERROR:    An error occurred: 'IAmRequest' object has no attribute 'apduInvokeID'
INFO:    Received Who Is Request from 192.168.1.10:57079
INFO:    I Am from device,2000
DEBUG:    Device device,2000 already in cache!
DEBUG:    Exploring Device info of device,2000
WARNING:    Abort PDU error while reading device properties: device,2000: segmentation-not-supported
INFO:    Reading device device,2000, object-identifier
ERROR:    Other error: name 'address' is not defined
INFO:    Reading objectList from device,2000...
ERROR:    exception: KeyError('device:2000')
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/app.py", line 899, in indication
    await helperFn(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 139, in do_IAmRequest
    await self.read_object_list(device_identifier=apdu.iAmDeviceIdentifier)
  File "/usr/bin/BACnetIOHandler.py", line 340, in read_object_list
    for obj_id in self.bacnet_device_dict[f"device:{device_identifier[1]}"][
```

Bepacom-Raalte, 2023-12-20:
> this was the log from when i conducted a scan/add device with YABE ,

I mean, could you export the scan from YABE?

azzey1, 2023-12-21:
yes, a couple of posts up

GravySeal, 2023-12-23:
Could you reinstall and try again? Hopefully this time it'll go a little better.
Seems there's an issue inside of the BACnet library I use where it fails to read a property of the device object.

azzey1, 2023-12-29:
Hi, How do i tell if the update has taken affect without a version number change?
Getting the following

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Bepacom EcoPanel BACnet/IP Interface Development Version
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2023.12.4
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[17:12:33] INFO: Running init-nginx.
[17:12:33] INFO: Running init-interface.
[17:12:33] INFO: Allowed interfaces for NGINX: 127.0.0.1/8 192.168.1.22/24 172.30.32.1/23 172.30.232.1/23
[17:12:33] INFO: Generated ingress configuration!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.1.22/24
objectIdentifier: 1999
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 5
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: DEBUG
defaultPriority: 8
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
[17:12:34] INFO: Running interface
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
DEBUG:    Application initialised
INFO:     Started server process [271]
INFO:     Waiting for application startup.
INFO:    I Am from device,2000
DEBUG:    Exploring Device info of device,2000
WARNING:    Abort PDU error while reading device properties: device,2000: segmentation-not-supported
INFO:    Reading device device,2000, object-identifier
ERROR:    Other error: name 'objid' is not defined
INFO:    Last resort for this device...
ERROR:    An error occurred: segmentation-not-supported
INFO:     Application startup complete.
INFO:     Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
WARNING:  Invalid HTTP request received.
[17:12:41] INFO: Running NGINX
2023/12/29 17:12:41 [notice] 281#281: using the "epoll" event method
2023/12/29 17:12:41 [notice] 281#281: nginx/1.24.0
2023/12/29 17:12:41 [notice] 281#281: OS: Linux 6.1.63-haos
2023/12/29 17:12:41 [notice] 281#281: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/29 17:12:41 [notice] 281#281: start worker processes
2023/12/29 17:12:41 [notice] 281#281: start worker process 326
```

GravySeal, 2023-12-29:
> How do i tell if the update has taken affect without a version number change?

The logging will be slightly different. When reinstalling the add-on, Home Assistant always builds the latest version from GitHub, so if a new update has been pushed and Home Assistant is aware of this update it'll install the last version.

Could you maybe try again? I changed the way the object list is retrieved in the alternative way, so hopefully it should get some data.

azzey1, 2023-12-29:
-----------------------------------------------------------
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2023.12.4
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[07:32:32] INFO: Running init-nginx.
[07:32:32] INFO: Running init-interface.
[07:32:32] INFO: Allowed interfaces for NGINX: 127.0.0.1/8 192.168.1.22/24 172.30.32.1/23 172.30.232.1/23
[07:32:32] INFO: Generated ingress configuration!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.1.22/24
objectIdentifier: 1999
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 4
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: DEBUG
defaultPriority: 8
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
[07:32:33] INFO: Running interface
s6-rc: info: service legacy-services successfully started
DEBUG:    Application initialised
INFO:     Started server process [271]
INFO:     Waiting for application startup.
INFO:    I Am from device,2000
DEBUG:    Exploring Device info of device,2000
WARNING:    Abort PDU error while reading device properties: device,2000: segmentation-not-supported
INFO:    Reading device device,2000, object-identifier
INFO:    Response device device,2000, object-identifier, <bacpypes3.primitivedata.Any object at 0x7f21f61e41a0>, <bacpypes3.apdu.ReadPropertyACK(ComplexAckPDU,1) instance at 0x7f21f61e4110>
INFO:    Reading device device,2000, object-type
INFO:    Response device device,2000, object-type, <bacpypes3.primitivedata.Any object at 0x7f21f61e42f0>, <bacpypes3.apdu.ReadPropertyACK(ComplexAckPDU,2) instance at 0x7f21f61e4230>
INFO:    Reading device device,2000, object-name
INFO:    Response device device,2000, object-name, <bacpypes3.primitivedata.Any object at 0x7f21f61e4260>, <bacpypes3.apdu.ReadPropertyACK(ComplexAckPDU,3) instance at 0x7f21f61e4080>
ERROR:    Other error: minimum length: 1 device,2000
INFO:    Reading objectList from device,2000...
ERROR:    exception: KeyError('objectList')
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/app.py", line 899, in indication
    await helperFn(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 139, in do_IAmRequest
    await self.read_object_list(device_identifier=apdu.iAmDeviceIdentifier)
  File "/usr/bin/BACnetIOHandler.py", line 351, in read_object_list
    for obj_id in self.bacnet_device_dict[f"device:{device_identifier[1]}"][
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'objectList'
ERROR:    An error occurred: 'IAmRequest' object has no attribute 'apduInvokeID'
INFO:     Application startup complete.
INFO:     Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
WARNING:  Invalid HTTP request received.
[07:32:40] INFO: Running NGINX
2023/12/30 07:32:40 [notice] 279#279: using the "epoll" event method
2023/12/30 07:32:40 [notice] 279#279: nginx/1.24.0
2023/12/30 07:32:40 [notice] 279#279: OS: Linux 6.1.63-haos
2023/12/30 07:32:40 [notice] 279#279: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/30 07:32:40 [notice] 279#279: start worker processes
2023/12/30 07:32:40 [notice] 279#279: start worker process 327
INFO:    Periodic object reading...
INFO:    Periodic object reading...
INFO:    Periodic object reading...
INFO:    Periodic object reading...
INFO:    Periodic object reading...
INFO:    Periodic object reading...
INFO:    Periodic object reading...

GravySeal, 2023-12-29:
I think I got pretty close right now. Unfortunately I don't have any real BACnet devices on hand, otherwise I could experiment with those. Could you try again for me please?

azzey1, 2023-12-29:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Bepacom EcoPanel BACnet/IP Interface Development Version
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2023.12.4
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[09:27:07] INFO: Running init-nginx.
[09:27:07] INFO: Running init-interface.
[09:27:07] INFO: Allowed interfaces for NGINX: 127.0.0.1/8 192.168.1.22/24 172.30.32.1/23 172.30.232.1/23
[09:27:07] INFO: Generated ingress configuration!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.1.22/24
objectIdentifier: 1999
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: DEBUG
defaultPriority: 8
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
[09:27:08] INFO: Running interface
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
DEBUG:    Application initialised
INFO:     Started server process [272]
INFO:     Waiting for application startup.
INFO:    I Am from device,2000
DEBUG:    Exploring Device info of device,2000
WARNING:    Abort PDU error while reading device properties: device,2000: segmentation-not-supported
INFO:    Reading device device,2000, object-identifier
INFO:    Response device device,2000, object-identifier, <bacpypes3.primitivedata.Any object at 0x7f49eb6f4410>, <bacpypes3.apdu.ReadPropertyACK(ComplexAckPDU,1) instance at 0x7f49eb6f4470>
INFO:    Reading device device,2000, object-type
INFO:    Response device device,2000, object-type, <bacpypes3.primitivedata.Any object at 0x7f49eb6f4350>, <bacpypes3.apdu.ReadPropertyACK(ComplexAckPDU,2) instance at 0x7f49eb6f46e0>
INFO:    Reading device device,2000, object-name
INFO:    Response device device,2000, object-name, <bacpypes3.primitivedata.Any object at 0x7f49eb6f43e0>, <bacpypes3.apdu.ReadPropertyACK(ComplexAckPDU,3) instance at 0x7f49eb6f42f0>
ERROR:    Other error: minimum length: 1 device,2000
INFO:    Reading objectList from device,2000...
ERROR:    exception: KeyError('objectList')
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/app.py", line 899, in indication
    await helperFn(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 139, in do_IAmRequest
    await self.read_object_list(device_identifier=apdu.iAmDeviceIdentifier)
  File "/usr/bin/BACnetIOHandler.py", line 351, in read_object_list
    for obj_id in self.bacnet_device_dict[f"device:{device_identifier[1]}"][
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'objectList'
ERROR:    An error occurred: 'IAmRequest' object has no attribute 'apduInvokeID'
INFO:     Application startup complete.
INFO:     Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
WARNING:  Invalid HTTP request received.
[09:27:14] INFO: Running NGINX
2023/12/30 09:27:14 [notice] 282#282: using the "epoll" event method
2023/12/30 09:27:14 [notice] 282#282: nginx/1.24.0
2023/12/30 09:27:14 [notice] 282#282: OS: Linux 6.1.63-haos
2023/12/30 09:27:14 [notice] 282#282: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/30 09:27:14 [notice] 282#282: start worker processes
2023/12/30 09:27:14 [notice] 282#282: start worker process 327
INFO:    Periodic object reading...
```

GravySeal, 2023-12-29:
Did you check for updates before reinstalling? This still behaves like the old version.

azzey1, 2023-12-29:
don't have a way to check for updates? how do i check?

azzey1, 2023-12-29:
uninstalled and re-installed

```
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2023.12.4
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[10:38:00] INFO: Running init-nginx.
[10:38:00] INFO: Running init-interface.
[10:38:00] INFO: Allowed interfaces for NGINX: 127.0.0.1/8 192.168.1.22/24 172.30.232.1/23 172.30.32.1/23
[10:38:00] INFO: Generated ingress configuration!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.1.22/24
objectIdentifier: 1999
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: DEBUG
defaultPriority: 8
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
[10:38:01] INFO: Running interface
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
DEBUG:    Application initialised
INFO:     Started server process [271]
INFO:     Waiting for application startup.
INFO:    I Am from device,2000
DEBUG:    Exploring Device info of device,2000
WARNING:    Abort PDU error while reading device properties: device,2000: segmentation-not-supported
ERROR:    exception: ValueError('minimum length: 1')
Traceback (most recent call last):
  File "/usr/bin/BACnetIOHandler.py", line 226, in read_device_props
    response = await self.read_property_multiple(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/service/object.py", line 543, in read_property_multiple
    response = await self.request(read_property_multiple_request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
bacpypes3.apdu.AbortPDU: segmentation-not-supported

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/app.py", line 899, in indication
    await helperFn(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 137, in do_IAmRequest
    await self.read_device_props(apdu=apdu)
  File "/usr/bin/BACnetIOHandler.py", line 238, in read_device_props
    await self.read_device_props_non_segmented(apdu)
  File "/usr/bin/BACnetIOHandler.py", line 268, in read_device_props_non_segmented
    response = await self.read_property(
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/service/object.py", line 146, in read_property
    property_value = response.propertyValue.cast_out(property_type)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/constructeddata.py", line 1751, in cast_out
    result = cls.decode(tag_list)
             ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/primitivedata.py", line 1490, in decode
    return cls(value)
           ^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/primitivedata.py", line 628, in __call__
    args = (new_type.cast(args[0]),)  # type: ignore[attr-defined]
            ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/primitivedata.py", line 1392, in cast
    raise ValueError(f"minimum length: {min_length}")
ValueError: minimum length: 1
ERROR:    An error occurred: 'IAmRequest' object has no attribute 'apduInvokeID'
INFO:     Application startup complete.
INFO:     Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
WARNING:  Invalid HTTP request received.
[10:38:07] INFO: Running NGINX
2023/12/30 10:38:07 [notice] 281#281: using the "epoll" event method
2023/12/30 10:38:07 [notice] 281#281: nginx/1.24.0
2023/12/30 10:38:07 [notice] 281#281: OS: Linux 6.1.63-haos
2023/12/30 10:38:07 [notice] 281#281: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/30 10:38:07 [notice] 281#281: start worker processes
2023/12/30 10:38:07 [notice] 281#281: start worker process 326
```

GravySeal, 2023-12-29:
> uninstalled and re-installed

Basically this. To ensure you get the latest version, in the add-on store click the "Check for updates" button in the top right. I just added an exception for valueError and it'll try to ignore it for now.

azzey1, 2023-12-30:
ok, that seams to have worked.
Bepacom EcoPanel BACnet/IP Interface Development Version
```
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.567169189453125
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.4239444732666
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.843292236328125
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.56719398498535
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.4237117767334
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.843412399291992
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.567216873168945
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.423486709594727
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.843528747558594
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.56723976135254
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.423269271850586
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.84364128112793
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.5672607421875
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.423057556152344
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.84375
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.56728172302246
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.422853469848633
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.843854904174805
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.567302703857422
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.42265510559082
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.843955993652344
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.56732177734375
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.422462463378906
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.84405517578125
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.567340850830078
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.42227554321289
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.84415054321289
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.567359924316406
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.422094345092773
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.844242095947266
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.5673770904541
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.421918869018555
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.844331741333008
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.567394256591797
[... 9 lines cut ...]
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.844501495361328
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.567428588867188
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.42142677307129
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.84458351135254
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.56744384765625
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.42127227783203
[... cut ...]
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.421123504638672
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
DEBUG:    COV: device,2000, analog-input,1, present-value, 29.844738006591797
DEBUG:    COV: device,2000, analog-input,1, status-flags, 
DEBUG:    COV: device,2000, analog-input,4, present-value, 28.567474365234375
DEBUG:    COV: device,2000, analog-input,4, status-flags, 
DEBUG:    COV: device,2000, analog-input,0, present-value, 30.420978546142578
DEBUG:    COV: device,2000, analog-input,0, status-flags, 
```

GravySeal, 2023-12-30:
Awesome! Do you see all objects you expect to see in the web UI?

azzey1, 2023-12-30:
yes it looks like they are all showing.

azzey1, 2023-12-30:
is there a reference manual to show how i setup the points for use and reference on dashboards?

azzey1, 2023-12-30:
i am getting this
```
ERROR:    Error while subscribing to device,2000, binary-value,263: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,264: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,265: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,266: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,267: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,268: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,269: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,270: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,271: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,272: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,273: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,274: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,275: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,276: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,277: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,278: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,279: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,280: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,281: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,282: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,283: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,284: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,285: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,286: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,287: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,288: resources: no-space-to-add-list-element
ERROR:    Error while subscribing to device,2000, binary-value,289: resources: no-space-to-add-list-element
```

GravySeal, 2023-12-30:
> is there a reference manual to show how i setup the points for use and reference on dashboards?

For entities to appear in Home Assistant, you'd have to install the custom integration.

https://github.com/Bepacom-Raalte/bepacom-custom_components/tree/main/custom_components/bacnet_interface

Installing the integration is as follows:

You create a folder named "custom_components" in your Home Assistant's /config folder.
In /config/custom_components you paste the "bacnet_interface" folder from the repository I linked.
Once this is done, restart Home Assistant and add this integration manually through Settings > Devices & Services > + Add Integration and look for BACnet.
When the integration opens, you'll have to type an IP address. Use 127.0.0.1.

> i am getting this

I think this is a limitation of the hardware. There's over 700 BACnet objects in your device, and subscribing to everything might be a bit too much. In the future I want to add some kind of whitelist so you only subscribe to selected objects. 

Currently, what you can try in the configuration of the add-on is set multiStateInput, multiStateOutput and analogOutput to false. This reduces a couple of subscriptions.

If you plan on running the add-on undisturbed for a long time, you can manually add the subscriptions you need in the web UI. Mind you that this will be reset once the add-on restarts. These subscriptions aren't saved persistently.

If somehow you can configure your device to not include all these extra empty objects, that should solve the problem as well.

Subscriptions aren't the only way to receive data. You can also set all subscriptions to false in the configuration and only get data through polling every 60 seconds. You can increase or decrease the time through the Update interval setting.

azzey1, 2023-12-30:
Thanks, new to HA.
The device unfortunately does not have setting to disable points to be visible. 
i will have a play

azzey1, 2023-12-30:
so subscriptions are for COV?
also testing the on/off control from HA, and the priority it writes too doesn't seam to change based on the configuration setting.
ie i have set Priority to 8 in configuration, and it is still writing at 16 (it did change value, however i have controller software writing to 14 so the present value state didn't change)

```
WARNING:    <PropertyIdentifier: status-flags> not supported
INFO:     None:0 - "POST /apiv1/device%3A2000/binaryOutput%3A7?deviceid=device:2000&objectid=binaryOutput:7&presentValue=1 HTTP/1.0" 200 OK
INFO:    Write result: inactive
2023/12/30 12:24:33 [info] 326#326: *17 client 127.0.0.1 closed keepalive connection
INFO:    Periodic object reading...
WARNING:    <PropertyIdentifier: status-flags> not supported
INFO:     None:0 - "POST /apiv1/device%3A2000/binaryOutput%3A7?deviceid=device:2000&objectid=binaryOutput:7&presentValue=1 HTTP/1.0" 200 OK
INFO:    Write result: inactive
2023/12/30 12:25:56 [info] 326#326: *19 client 127.0.0.1 closed keepalive connection
INFO:    Periodic object reading...
WARNING:    <PropertyIdentifier: status-flags> not supported
INFO:     None:0 - "POST /apiv1/device%3A2000/binaryOutput%3A7?deviceid=device:2000&objectid=binaryOutput:7&presentValue=1 HTTP/1.0" 200 OK
INFO:    Write result: inactive
2023/12/30 12:26:41 [info] 326#326: *21 client 127.0.0.1 closed keepalive connection 
```

GravySeal, 2023-12-30:
> so subscriptions are for COV?

This is correct yes.

> also testing the on/off control from HA, and the priority it writes too doesn't seam to change based on the configuration setting.

You're right, thank you for letting me know. I'll make a fix for this as well.

GravySeal, 2023-12-30:
Would you mind testing the priority for me? I just pushed a possible fix.

azzey1, 2023-12-30:
Have just tested, and works. Thanks!

azzey1, 2023-12-30:
let me know if there is anything else you would like tested.

GravySeal, 2023-12-30:
> Have just tested, and works. Thanks!

Great, glad it works! 

> let me know if there is anything else you would like tested.

For sure, if there's any issues just let me know as well. I'll push these updates to the main version sometime early next year when I have some real devices to test with again.

Bepacom-Raalte, 2024-01-10:
Pushed to the regular version with 1.3.0.

closed 2024-01-10 by Bepacom-Raalte