Bepacom-Raalte/bepacom-HA-Addons #32: When I enter an address for the BBMD, the service fails to start.

rriemerHass, 2023-12-24:
Here is a snippet of the logs:  I have tried with a bbmd address like x.x.x.x and x.x.x.x/y

[11:18:01] INFO: Running interface
ERROR:    An error occurred: 
Traceback (most recent call last):
  File "/usr/bin/main.py", line 261, in <module>
    asyncio.run(main())
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 664, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/bin/main.py", line 188, in main
    app = BACnetIOHandler(
          ^^^^^^^^^^^^^^^^
  File "/usr/bin/BACnetIOHandler.py", line 41, in __init__
    self.register(addr=Address(foreign_ip), ttl=ttl)
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/ipv4/app.py", line 180, in register
    self.foreign.register(addr, ttl)
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/ipv4/service.py", line 668, in register
    if ttl <= 0:
       ^^^^^^^^
TypeError: '<=' not supported between instances of 'str' and 'int'

GravySeal, 2023-12-24:
Could you try again with the development version?
Please uninstall and check for updates first if you were already using the development version. 
I see that ttl wasn't being interpreted as integer so I just pushed a fix for that.

rriemerHass, 2023-12-25:
Seems to have corrected the issue.  Thanks, Merry Christmas!

GravySeal, 2023-12-26:
Great, glad to hear! I'll push an update to the regular version sometime next year. Merry (belated) Christmas to you as well.

Bepacom-Raalte, 2024-01-10:
Pushed to the regular version with 1.3.0.

closed 2024-01-10 by Bepacom-Raalte