Bepacom-Raalte/bepacom-HA-Addons #33: Latest update fails to start - last updated at 13:00 GMT-6.

rriemerHass, 2024-01-07:
Log follows:

s6-rc: info: service interface: starting
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[14:51:02] INFO: Running interface
DEBUG:    Application initialised
ERROR:    An uncaught error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 684, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/bin/main.py", line 199, in main
    object_manager = ObjectManager(app=app, entity_list=options['entity_list'], api_token=token)
                                                        ~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'entity_list'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 84, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/ipv4/__init__.py", line 155, in set_local_transport_protocol
    transport, protocol = task.result()
                          ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1437, in create_datagram_endpoint
    await waiter
asyncio.exceptions.CancelledError
ERROR:    An uncaught error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 684, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/bin/main.py", line 199, in main
    object_manager = ObjectManager(app=app, entity_list=options['entity_list'], api_token=token)
                                                        ~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'entity_list'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 84, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/ipv4/__init__.py", line 186, in set_broadcast_transport_protocol
    transport, protocol = task.result()
                          ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1437, in create_datagram_endpoint
    await waiter
asyncio.exceptions.CancelledError
Traceback (most recent call last):
  File "/usr/bin/main.py", line 270, in <module>
    asyncio.run(main())
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 684, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/bin/main.py", line 199, in main
    object_manager = ObjectManager(app=app, entity_list=options['entity_list'], api_token=token)
                                                        ~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'entity_list'
[14:51:13] INFO: Service BACnet Interface exited with code 1 (by signal 0)
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service nginx: stopping
[14:51:13] INFO: Service NGINX exited with code 256 (by signal 15)
s6-rc: info: service nginx successfully stopped
s6-rc: info: service init-nginx: stopping
s6-rc: info: service interface: stopping
s6-rc: info: service interface successfully stopped
s6-rc: info: service init-interface: stopping
s6-rc: info: service init-nginx successfully stopped
s6-rc: info: service init-interface successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

rriemerHass, 2024-01-07:
This is with the latest update of Home Assistant as well.

GravySeal, 2024-01-07:
Should be fixed, please try again!

rriemerHass, 2024-01-07:
Uninstalled, updated addons, reinstalled, but still:

s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[15:26:58] INFO: Running init-nginx.
[15:26:58] INFO: Running init-interface.
[15:26:58] INFO: Allowed interfaces for NGINX: 127.0.0.1/8 192.168.13.109/24 172.30.32.1/23 172.17.0.1/16
[15:26:58] INFO: Generated ingress configuration!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: House
address: 192.168.13.109/24
objectIdentifier: 11
maxApduLengthAccepted: 1476
segmentation: noSegmentation
vendorIdentifier: 15
foreignBBMD: 192.168.100
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: WARNING
defaultPriority: 15
updateInterval: 60
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[15:26:59] INFO: Running interface
ERROR:    An uncaught error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 684, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/bin/main.py", line 199, in main
    object_manager = ObjectManager(app=app, entity_list=options['entity_list'], api_token=token)
                                                        ~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'entity_list'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 84, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/bacpypes3/ipv4/__init__.py", line 155, in set_local_transport_protocol
    transport, protocol = task.result()
                          ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1437, in create_datagram_endpoint
    await waiter
asyncio.exceptions.CancelledError
Traceback (most recent call last):
  File "/usr/bin/main.py", line 270, in <module>
    asyncio.run(main())
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 684, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/bin/main.py", line 199, in main
    object_manager = ObjectManager(app=app, entity_list=options['entity_list'], api_token=token)
                                                        ~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'entity_list'
[15:27:10] INFO: Service BACnet Interface exited with code 1 (by signal 0)
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service nginx: stopping
[15:27:11] INFO: Service NGINX exited with code 256 (by signal 15)
s6-rc: info: service nginx successfully stopped
s6-rc: info: service init-nginx: stopping
s6-rc: info: service interface: stopping
s6-rc: info: service interface successfully stopped
s6-rc: info: service init-interface: stopping
s6-rc: info: service init-nginx successfully stopped
s6-rc: info: service init-interface successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

GravySeal, 2024-01-07:
Don't forget to click "Check for updates" first in the add-on store!

rriemerHass, 2024-01-08:
I did, but either it did not update correctly, or the issue still exists.  How can I be sure it updated correctly?  I uninstall, restart HA, refreshed addons, then installed.

Bepacom-Raalte, 2024-01-08:
Seems I overlooked that Home Assistant doesn't automatically generate a new configuration option when one gets added. I changed the code again so that now it shouldn't matter. If you check for updates again and then select rebuild or you reinstall, it should solve the issue.

rriemerHass, 2024-01-10:
Yup, that has done it!  Running again.  Thank You!

closed 2024-01-10 by rriemerHass