Bepacom-Raalte/bepacom-HA-Addons #38: Not all entities are being created

CV-Private, 2024-03-13:
After (re)installing the Bepacom Add-On, I found that only 63 Entities had been created, even though the WebUI had already completed scanning my GTW-21 BACnet device and found all 1279 objects.

[image]

I then restarted HA, and after the restart, all 1279 entities are there:

[image]

Why was a restart needed to get all entities?
By the way, I saw this behavior before in previous installations, but didn't take the time for taking screenshots.

Bepacom-Raalte, 2024-03-14:
It means the integration connects before everything has been scanned in the BACnet network.
It's a bit of a limitation of the integration, as there is currently no dynamic generation of entities.
The integration only creates entities for the data it receives on it's first update.

CV-Private, 2024-03-24:
Until dynamic generation of entities is possible, a workaround could be to postpone the startup of the Integration until the Add-On has finished scanning the entire object tree. While this might cause some delay, it should still be understandable for the user if the Integration would display a message at startup, saying something like: "_The Intergration will start as soon as all BACnet devices have been scanned. Please wait._"

closed 2024-03-27 by Bepacom-Raalte