Bepacom-Raalte/bepacom-HA-Addons #46: The entity does not consider the Resolution of a BACnet object

CV-Private, 2024-03-28:
The automatically generated Entity does not consider the Resolution of a BACnet object.

Here's an example with my GTW-21 device: Analog Value AV1673 _parZoneMode-3_ is a Mode switch that has only three valid states: 0, 1 and 2. This is reflected in the object properties Min Pres Value 0, Max Pres Value 2, **Resolution 1**, as can be seen in Yabe:

[image]

The automatically generated Entity, however, allows to increment the value in 0.1 steps:

[image]

It seems that the BACnet device rejects that invalid write command - that would at least be my understanding of the orange bullet in the log. Setting the property to a valid value returns a blue bullet in the log:

[image]

Despite the rejection of the first command, the Entity was displaying the value as 1.1 without showing any error.

GravySeal, 2024-03-31:
You're right about resolution, I'll include it in the next add-on release. After that, it'll be added to the integration.

gioassio89, 2024-04-15:
Hi,
seems that there is some in the definition of the resolution for analog writable:

[image]

[image]

GravySeal, 2024-04-17:
The integration looks whether resolution exists, if it doesn't it checks of CoV increment exists, and if neither are there it'll use a resolution of 1. Maybe it's missing those BACnet properties?

CV-Private, 2024-04-18:
The resolution is now being considered in v1.4.1b6 and works fine with the GTW-21. For example, the AV1673 value is now changing in integer steps when I click the up/down buttons.

Would be great, though, if the Integration would also adjust the decimal places according to the resolution. Currently, the AV1673 value is shown as 1.0 in the Dashboard, even though the resolution is 1. Could it be shown as an integer number instead?

[image]

Bepacom-Raalte, 2024-04-19:
Could you try v0.2.1b3? Should be fixed in this version. It'll check based on the step, which is based on resolution or CoV increment. When the resolution is one or greater, the value will be integer instead of floating point.

CV-Private, 2024-04-19:
With v0.2.1b3, I do see progress in the editable fields.

However, in the non-editable sensor readings (sorry - I don't have a better name for that), the decimal places are not yet corresponding to the resolution of the object, which I read out with Yabe:

[image]

Bepacom-Raalte, 2024-04-19:
Right, sensor is different from number. I've changed the sensor to be rounded based on resolution or CoV increment in v0.2.1b4. Please try it.

CV-Private, 2024-04-19:
Great, works fine in v0.2.1b4, also with AI460 that has a resolution of 0.001:

[image]

Bepacom-Raalte, 2024-04-19:
I think this would be something Home Assistant related. If all the other decimals are corresponding to their resolution or CoV increment, then it's something out of our control.

CV-Private, 2024-04-19:
Works fine now. Thanks! :-)

closed 2024-04-19 by CV-Private