Bepacom-Raalte/bepacom-HA-Addons #47: No recognition of units

SQ9MDD, 2024-03-31:
I have set up some bacnet objects that come with measurements:
temperature - works fine
wind speed - works correctly
atmospheric pressure (hPa) - I read the value correctly and the unit: QERR_CLASS_PROPERTY: QUERR_CODE_UNKNOWN_PROPERTY
battery voltage - I read the value correctly and the unit: QERR_CLASS_PROPERTY: QUERR_CODE_UNKNOWN_PROPERTY

In this case, the unit resulting from the unit in home assistant or "noUnits" should be used.
This is well described here:
https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/basetypes.py

However, if I set up a switch object, then all values of all objects change to QERR_CLASS_PROPERTY: QUERR_CODE_UNKNOWN_PROPERTY

Translated with DeepL.com (free version)
[image]
[image]
[image]

SQ9MDD, 2024-03-31:
Below is a screenshot and a file with logged errors:

[image]

[err.txt](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/14814562/err.txt)

GravySeal, 2024-03-31:
Hey @SQ9MDD,

Thanks for your detailed report! You're right about the noUnits property.
If the add-on can't figure out the units, it should default to noUnits.
I'll add this in the next release, along with (milli)volt and hectopascal. 

As for the switch problem, could you please download a full log through the web UI? This will give me some more context.

[image]

SQ9MDD, 2024-03-31:
Hi.

You were quick to respond. Below is the log you asked for. I'm wondering but these errors I'm seeing in Enteliweb are more of its response to the response from bacnet and not necessarily the error itself from the integration.
But I could be wrong.

[Uploading bacnet_addon_logs-err.txt…]()

GravySeal, 2024-03-31:
Seems like there's an issue internally in the BACpypes3 library. I'll increase the logging in the next update to see whether we can remedy it. 
Seems like during a read multiple request, the library fails to handle it correctly. 
I don't know what property it tried to read, might be worth it to find that out as well.

SQ9MDD, 2024-03-31:
Yes but this only occurs if I add some switches. If there are none it works. I would be very happy to help with testing. How can I help ?

[image]

GravySeal, 2024-03-31:
Is it being read by an application on your computer? 
If so, maybe you can check the BACnet traffic through Wireshark to see what's happening.
Otherwise we'll have to wait until add-on version v1.4.1b2 to see more detailed logging for this issue.

SQ9MDD, 2024-03-31:
I'm temporarily away from home but when I get back I'll turn on wireshark and try to catch this communication. Unfortunately this eweb machine, is not accessible from the outside with a remote desktop.
Anyway this project has a lot of potential, thanks for your work.

GravySeal, 2024-03-31:
Of course, thank you for helping me out as well!

GravySeal, 2024-04-01:
Hey, 

thank you for your recording!
It might be crashing because the switch object doesn't have a priority array. 
Something seems to be going wrong in the library function and instead of just putting an error in the response, it just throws an error in the program.
What it looks like to me is the BACpypes3 code misses that value is None and tries to work with it [here](https://github.com/JoelBender/BACpypes3/blob/1e6e7d4089f760d7cdfcf4d1e0aab57ac663a0fa/bacpypes3/constructeddata.py#L1400).
I'll try to fix it, but you'll have to test for me if it works or not.
If your program expects a priority array, I'll have to add it later.

SQ9MDD, 2024-04-01:
I did some more tests. When I turn off segmentation, basically you can add switches, and only they are described by the error code "operation problem" interestingly enough from the bacnet posiom I can control them.

But still the main problem is the lack of these units.

I'm ready for more tests :)

[image]

SQ9MDD, 2024-04-01:
...Ok it is not only nosegmentation but in combination with it MAX-APDU 120

[image]

GravySeal, 2024-04-01:
Can you try dev version v1.4.1b2? I just pushed an update with possible fixes. Might have to check for updates in the add-on store to find it.

SQ9MDD, 2024-04-01:
As for the subject of this submission, it's resolved !!!
All entities that have units unrecognized are simply without units, which at this stage is great because it does not generate errors.

As for the switch, it spoils everything on standard settings. If I set noSegmentation together with APD to 120 then only the switch is broken when reading.
I can control it from bacNet :)

Great job!!!

[image]

GravySeal, 2024-04-01:
Great, glad the main issue is solved!
Could you please provide me with add-on logs for the second issue when you set it to segmentedBoth? I'd love to solve that issue as well.

SQ9MDD, 2024-04-01:
If you just feel like repairing then I am available.
Below is the view from the navigator, the log from the app and a snapshot from wireshark.

[bacnet_addon_logs-actual.txt](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/14826335/bacnet_addon_logs-actual.txt)

[bacnet-traffic-actual.pcapng.txt](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/14826336/bacnet-traffic-actual.pcapng.txt)

[image]

GravySeal, 2024-04-01:
Thank you for the logs. I submitted a pull request to the BACpypes3 library, hopefully it can be solved that way.

SQ9MDD, 2024-04-01:
Thank you for your patience, I think the work you have done allows you to close this issue. And now we are waiting for backpypes :) It is necessary to bump that thread ?

GravySeal, 2024-04-01:
I'll close this issue yeah, I'll let you know if the fix has been accepted, so no need to do anything else. Thank you for your help!

closed 2024-04-01 by GravySeal