Bepacom-Raalte/bepacom-HA-Addons #48: Inability to subscribe to analog values object due to the "optional-functionality-not-supported"

aphilgr1, 2024-04-01:
I am using your integration to control 4 bacnet devices. Devices 77000 and 77001 are heat pumps with Carel controllers. Device 17800 is a loytec controller that runs our BMS and device 77003, is Philips - Dynalite PDDEG with bacnet support. 

On the Philips device, temperature readings work well, but commands for presets and set points, can pass from HA to dynet (Dynalte protocol) but not the other way around. It seems there is only one-way communication, since values changed from dynet, are shown in YABE, but do not pass-htrough to HA. The log file, indicates inability to subscribe to analog values for device 77003 due to the "optional-functionality-not-supported" object.

This is how the Philips device appears in YABE

[image]

These are the properties of the data-poins with analogue values that cannot pass from dynet to bacnet

[image]

this is my log file, downloaded from the integration WebUI

[bacnet_addon_logs (4).txt](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/14824973/bacnet_addon_logs.4.txt)

Thanks in advance and congratulations on the integration. I love it!

Bepacom-Raalte, 2024-04-02:
Hey,

Thank you for using our add-on and integration!
The "optional-functionality-not-supported" message means that this BACnet device does not support Change of Value subscriptions.
Seems there are some more issues with subscribing to certain objects.
The reason Home Assistant isn't getting updated quickly is because the subscription for that object likely failed. This means that the add-on isn't getting notified if it's value changes.

If you're feeling bold, you can try the dev version v1.4.1b4.
In this version, I'm experimenting with configurations for specific devices. 
An example below: 

```yaml
devices_setup:
  - deviceID: device:77003
    CoV_lifetime: 600 #600 seconds until objects get resubscribed automatically.
    CoV_list: [] #empty so it won't subscribe to anything
    quick_poll_rate: 5 #every object under quick_poll_rate will be read every 5 seconds. keyword "all" can't be used here.
    quick_poll_list: 
      - analogValue:1839
      - analogValue:1940 #any other objects you want to be polled quickly.
    slow_poll_rate: 600
    slow_poll_list:
      - all #every object will be read every 600 seconds.
  - deviceID: device:77001
    CoV_lifetime: 600
    CoV_list: 
      - all
    quick_poll_rate: 5 
    quick_poll_list: []
    slow_poll_rate: 600
    slow_poll_list:
      - all
  - deviceID: device:77000
    CoV_lifetime: 600
    CoV_list: 
      - all
    quick_poll_rate: 5 
    quick_poll_list: []
    slow_poll_rate: 600
    slow_poll_list:
      - all
  - deviceID: all #any remaining devices not mentioned will use these settings.
    CoV_lifetime: 600
    CoV_list: 
      - all
    quick_poll_rate: 5 
    quick_poll_list: []
    slow_poll_rate: 600
    slow_poll_list:
      - all
```

You could use quick polling for any object you want to be updated quickly, so might be worth a try.

aphilgr1, 2024-04-02:
Thanks for the support and the quick reaction. 

My problem at the moment is, that since beta2 I get high CPU usage and the system becomes unresponsive. After a while, the web UI would not load. Trying to reload the interface, I get message "initializing" and after a while, "failed setup, will retry". I had to revert to beta1 (from restore backup) to make it work again

aphilgr1, 2024-04-03:
Still waiting for v1.4.1b5 to appear as update! Looking forward to try it

Bepacom-Raalte, 2024-04-03:
If you press "Check for updates" in the add-on store, it should appear on the add-on page.

aphilgr1, 2024-04-03:
It works!!! Super!!!
Here is the latest log 
[bacnet_addon_logs (7).txt](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/14851975/bacnet_addon_logs.7.txt)

Bepacom-Raalte, 2024-04-03:
Awesome! So now you can try to configure everything to be as responsive as your prefer with a combination of polling and subscribing.

aphilgr1, 2024-04-23:
Hello again,

I am facing some issue. I added 3 datapoints on my loytec controller and then I lost most of the datapoints in bepacom

this is what I get in YABE
[image]

and this I get in bepacom
[image]

resulting in this
[image]

My log file
[bacnet_addon_logs (14).txt](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/15072891/bacnet_addon_logs.14.txt)

GravySeal, 2024-04-23:
This might be an integration issue, which version are you currently on?
Could you share Home Assistant (not the add-on) logs?
What happens if you revert to version v0.2.0?

aphilgr1, 2024-04-23:
It all started when I added 3 more datapoints on my loytec bacnet server. My system automatically uptated to beta6, I thought that this could be the issue, reverted (from full vm restore) to beta5 but the problem is exactly the same. The problem is that my installation runs on a hotel that is already open, and redeploying the integration creates havoc! There must be something to do with the loytec server and the fact that only datapoints from 1000 upwards are recognized.

You mean this log?

[97683af0_bacnetinterface_dev_2024-04-23T19-31-05.922Z.log](https://github.com/Bepacom-Raalte/bepacom-HA-Addons/files/15082111/97683af0_bacnetinterface_dev_2024-04-23T19-31-05.922Z.log)

GravySeal, 2024-04-23:
I mean the log at Settings > System > Logs. These should tell me at least a little about what's happening with the integration.
Do the points that are still there update?
What type of BACnet objects did you add?

aphilgr1, 2024-04-24:
The objects were binary values. I reverted my Loytec configuration and the entities came back to life. Just without the 3 new data points. That means no access to problem log files with the problem. For now I am good, since the hotel is open and there is not much room for experiment. Thanks for your immediate response.