Bepacom-Raalte/bepacom-HA-Addons #49: Request to add units

SQ9MDD, 2024-04-04:
I want to transfer some readings from homeassistant to bacnet. 
I am missing units:

µg/m³ for air quality parameters,
m³ for water consumption measurements

Thanks a lot.

GravySeal, 2024-04-04:
Sure thing, I'll add them next beta!

gioassio89, 2024-04-15:
Hi,
seems miss also energy units: kWh
time units: s min h, 
this integration is fantastic! 
Thanks a lot!

SQ9MDD, 2024-04-16:
kWh is ok, also Celsius, lx, V, W, we are waiting for the rest of them, check this out:

@GravySeal any plans for update ?

[image]

Bepacom-Raalte, 2024-04-17:
Hello,

I'll add seconds, minutes and hours as engineering units as well. I'd been hoping on a little more feedback for the beta, as there's quite a few changes to the inner workings. I think I'll push update 1.4.1b6 tonight including the units added.

gioassio89, 2024-04-17:
Hi,
HA side i cannot see points that have W KWh dont have untits:
[image]
But the device that export point have it, % °C have no problem:
[image]
I do something wrong?

Bepacom-Raalte, 2024-04-17:
This is from a BACnet device to Home Assistant, right? Currently the integration supports only 3 units, so this is an integration issue and not the add-on. If you have a list of units you want supported in Home Assistant, you can make an issue on the [integration GitHub page](https://github.com/Bepacom-Raalte/Bepacom-BACnet-IP-Integration/issues) so I can add them there.

gioassio89, 2024-04-17:
sorry i will make an issue in the correct place! thanks a lot!

closed 2024-04-19 by SQ9MDD