Bepacom-Raalte/bepacom-HA-Addons #50: The Add-On did not restore the previous operating state after the BACnet device was power-cycled.

CV-Private, 2024-04-04:
I had to turn the GTW-21 off and on, and I found that the Add-On did not re-subscribe to CoV after the BACnet device was power-cycled. 

CoV subscriptions and polling were active before the power-cycle, but not thereafter. I see in Yabe that there are no active subscriptions, even though the Add-On has subscriptions in the list.

I'd propose to add a recovery mechanism that restores the previous operating state if the connections to the BACnet device was interrupted.

[image]

GravySeal, 2024-04-04:
There's plans to make some sort of a heartbeat functionality to poll whether a device is still there, but for now it doesn't have a priority.
A way to work around this issue is having the CoV heartbeat shorter, like 120 seconds for example. This is a similar way to how Yabe does it.

CV-Private, 2024-04-04:
Hmm.. can I configure the heartbeat in the current Add-On? I didn't find anything in your documentation about that option...

Bepacom-Raalte, 2024-04-05:
There currently is no heartbeat, just something I want to implement later.

As for resubscribing, restarting Home Assistant does nothing to the add-on. Only rebooting the host device will also restart the add-on. Instead, you can decrease the lifetime of a CoV subscription from 28800 to 120 seconds. This should resubscribe. If it doesn't, feel free to leave some logs so I can take a look at what's happening.

GravySeal, 2024-09-07:
The add-on should now restore CoV subscriptions after the device lost connection. This will be done through resubscribing when the lifetime of the subscription has passed.

closed 2024-09-07 by GravySeal