Bepacom-Raalte/bepacom-HA-Addons #52: No CoV re-subscription after the connection to the BACnet device was interrupted

CV-Private, 2024-04-21:
After I disconnected and re-connected the GTW-21 from the network, the previously active CoV subscriptions were not re-established. 

Only the Quick-Poll values are being updated in the Integration, but there's no more update of the CoV values since the network connection was down, which is meanwhile more than two hours ago:

[image]

[image]

I think this was already described as part of another issue, but hasn't been addressed yet, therefore I'm opening it as a new report.

GravySeal, 2024-04-21:
Yes, thank you for making this a separate issue. Once I have more time I'll look into a solution for this problem. I've been testing around a little bit but unfortunately haven't found an easy fix yet.

GravySeal, 2024-08-18:
In v1.5.1b1 it should resubscribe now after the CoV lifetime has passed.

closed 2024-09-07 by GravySeal