Bepacom-Raalte/bepacom-HA-Addons #58: Daikin D-BACS

BradleyFord, 2024-04-27:
**Describe the bug**
I can see data streaming into YABE and the Daikin D-BACS tool; add-on is showing various errors subscribing to data.

But I can see data in the add-on interface:
[image]

When using the integration nothing is shown
[image]

1. I have tried building a few times
2. I have checked D-BACS to ensure COV is enabled
3. I have changed the entity naming style a few times. (Which explains those very tiny "blips" in the HA chart)

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

`[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-value,5644: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-value,5645: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-value,5646: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-value,5648: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-input,5652: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, analog-value,5654: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, analog-input,5655: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-output,5656: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-input,5657: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-output,5658: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-input,5659: 
[02:11:00]|ERROR    |BACnetIOHandler.py->subscription_task                   : Cancelling subscription task: device,0, binary-input,5660: 
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Bepacom BACnet/IP Interface Development Version
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet/IP devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.4.1b7
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.2  (amd64 / qemux86-64)
 Home Assistant Core: 2024.4.3
 Home Assistant Supervisor: 2024.04.4
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[02:11:12] INFO: Running init-nginx.
[02:11:12] INFO: Running init-interface.
[02:11:12] INFO: Allowed addresses for NGINX: 127.0.0.1 192.168.1.222 172.30.232.1 172.30.32.1
Using enp0s3 as address: 192.168.1.222/24
[02:11:12] INFO: Generated ingress configuration successfully!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: D-BACS
address: 192.168.1.222/24
objectIdentifier: 420
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: WARNING
defaultPriority: 15
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
[02:11:13] INFO: Running interface
s6-rc: info: service legacy-services successfully started
[02:11:25] INFO: Running NGINX
[02:11:25]|WARNING  |h11_impl.py       ->handle_events                       : Invalid HTTP request received.
2024/04/28 02:11:25 [notice] 301#301: using the "epoll" event method
2024/04/28 02:11:25 [notice] 301#301: nginx/1.24.0
2024/04/28 02:11:25 [notice] 301#301: OS: Linux 6.6.25-haos
2024/04/28 02:11:25 [notice] 301#301: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/04/28 02:11:25 [notice] 301#301: start worker processes
2024/04/28 02:11:25 [notice] 301#301: start worker process 362
[02:11:31]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,274: services: other
[02:11:31]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,275: services: other
[02:11:34]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,530: services: other
[02:11:34]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,531: services: other
[02:11:37]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,786: services: other
[02:11:37]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,787: services: other
[02:11:40]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,1042: services: other
[02:11:40]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,1043: services: other
[02:11:42]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,4370: services: other
[02:11:42]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,4371: services: other
[02:11:45]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,4626: services: other
[02:11:45]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,4627: services: other
[02:11:48]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,4882: services: other
[02:11:48]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,4883: services: other
[02:11:51]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,5138: services: other
[02:11:51]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,5139: services: other
[02:11:54]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,5394: services: other
[02:11:54]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,5395: services: other
2024/04/28 02:11:55 [info] 362#362: *1 client 127.0.0.1 closed keepalive connection
[02:11:57]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,5650: services: other
[02:11:57]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,0, accumulator,5651: services: other
2024/04/28 02:12:31 [info] 362#362: *5 client 127.0.0.1 closed keepalive connection
2024/04/28 02:12:31 [info] 362#362: *7 client closed connection while waiting for request, client: 127.0.0.1, server: 0.0.0.0:8099
2024/04/28 02:12:34 [info] 362#362: *8 client 127.0.0.1 closed keepalive connection
2024/04/28 02:12:56 [info] 362#362: *12 client 127.0.0.1 closed keepalive connection
2024/04/28 02:12:56 [info] 362#362: *14 client closed connection while waiting for request, client: 127.0.0.1, server: 0.0.0.0:8099
2024/04/28 02:12:58 [info] 362#362: *15 client 127.0.0.1 closed keepalive connection
2024/04/28 02:13:58 [info] 362#362: *19 client 127.0.0.1 closed keepalive connection
2024/04/28 02:14:58 [info] 362#362: *23 client 127.0.0.1 closed keepalive connection`

BradleyFord, 2024-04-28:
Moving to the HA Integration

GravySeal, 2024-04-28:
Hey BradleyFord,

I think the root cause is similar to [issue #19](https://github.com/Bepacom-Raalte/Bepacom-BACnet-IP-Integration/issues/19) at the integration GitHub. 

When you open the add-on web UI, there's a button to download logs. Could you download these logs for me and paste them here?
After this, I'm curious about the error that appeared 40 times in your screenshot from the Home Assistant logs. Could you maybe paste those logs here?
The only CoV objects failing from your screenshots appear to be accumulator objects. It doesn't seem other subscriptions have failed. I think receiving the data is alright, but sending it out is probably the culprit.

BradleyFord, 2024-04-28:
I think we close this, and I can join #19 as everything looks exactly the same in terms of error.

Those accumulator objects are for totalizing the power and gas consumption, but at the moment they are not moving above 0 as I dont have the optional equipment and the internal software is not setup, so this could be why they are failing COV.

It seems when the Add-on starts up it collects all the data ok...but then never updates again even after refresh the addon UI.

closed 2024-05-01 by BradleyFord