Bepacom-Raalte/bepacom-HA-Addons #61: FYI: Custom Climate Template

BradleyFord, 2024-05-03:
Not a bug.... I just want this to be recorded somewhere as it looks like a perfect companion for Bepacom

One of the challenges with Bepacom from my experience is mapping the entities that it creates to a HA Climate template.. which isnt possible.
But it seems an old climate template addition has been revived and can be added via HACS.
https://github.com/litinoveweedle/hass-template-climate

I have been setting this up over the last couple of days, at the moment I have managed to get it reading everything perfectly well. But I am still stuck getting it to write back, although I know the issue is somewhere with my configuration and the way Bepacom is handling the multistates.

Anyway, hope this is of benefit for everyone.

(Greyseal you can close this issue, but atleast if people search they will come across it)

GravySeal, 2024-05-03:
Awesome, thanks for sharing!

If you're having difficulty writing, maybe I can help out.
Feel free to share the config and I'll see if anything sticks out.

BradleyFord, 2024-05-07:
Sorry it took me a while to get back, I needed to get the fence finished before rainy season started... quite literaly I did the last coat of paint yesterday and it started raining today!!

Everything below seems to be working ok, in terms of reading and representing all of the data inside HA correctly. Except the write back...
There is one odd point with the Daikins, ON/OFF is seperate from the mode. So you can see in the HVAC mode template I have it first checking if the unit is turned off, then elseif the actual mode as it should be running is OFF is false.

Going through this process what I do not is different is the numbers in the Bepacom States are different from directly looking at the D-BACS interface. How are the numbers derived in Bepacom?

Also you will see that so far I am only trying to set fan speed, the plan is to get that working and then replicate to the other aspects, then finally to the other AC units.

`climate:
  - platform: climate_template
    unique_id: l2_ac_office
    name: L2_AC_Office
    hvac_modes:
      - "dry"
      - "off"
      - "cool"
      - "fan_only"
    swing_modes:
      - "off"
      - "vertical"
    fan_modes:
      - "auto"
      - "low"
      - "medium"
      - "high"
    min_temp: 24
    max_temp: 30

    current_temperature_template: "{{ states('sensor.bacnet_roomtemp_020') }}"
    target_temperature_template: "{{ states('number.bacnet_tempadjust_020') }}"
    hvac_mode_template: "{% if states('binary_sensor.bacnet_thermostatus_020') == 'off' %}off{% elif states('sensor.bacnet_airconmodestatus_020') == '1' %}cool{% elif  states('sensor.bacnet_airconmodestatus_020') == '3'%}fan{% elif  states('sensor.bacnet_airconmodestatus_020') == '5'%}dry{% else %}off{% endif %}"
    fan_mode_template: "{% if states('sensor.bacnet_airflowratestatus_020') == '4' %}auto{% elif  states('sensor.bacnet_airflowratestatus_020') == '1'%}low{% elif  states('sensor.bacnet_airflowratestatus_020') == '3'%}medium{% elif  states('sensor.bacnet_airflowratestatus_020') == '2'%}high{% else %}off{% endif %}"
    
    set_fan_mode:
      - service: climate.set_fan_mode
        target:
          entity_id: select.bacnet_airflowratecommand_020
        data:
          #fan_mode: "{{ fan_mode }}"
          #fan_mode: >-
          #  {% if fan_mode == '1' %}
          #    low
          #  {% elif fan_mode == '2' %}
          #    medium
          #  {% elif fan_mode == '3' %}
          #    high
          #  {% elif fan_mode == '4' %}
          #    auto
          #  {% endif %}
          fan_mode: >-
            {% if fan_mode == 'low' %}
              1
            {% elif fan_mode == 'medium' %}
              3
            {% elif fan_mode == 'high' %}
              2
            {% elif fan_mode == 'auto' %}
              4
            {% endif %}

GravySeal, 2024-05-07:
> quite literaly I did the last coat of paint yesterday and it started raining today!!

Right on time!

I tried my hand at a little test setup and got the following working:
```
climate:
  - platform: climate_template
    name: Bedroom Aircon
    hvac_modes:
      - "off"
      - "cool"
      - "heat"
    min_temp: 16
    max_temp: 30
    target_temperature_template: "{{ states('number.roomcontroller_simulator_setpoint_1') }}"
    current_temperature_template: "{{ states('sensor.roomcontroller_simulator_temperature_indoor') }}"
    hvac_mode_template: "{% if states('select.roomcontroller_simulator_state') == 'Stop' %}off{% elif states('select.roomcontroller_simulator_state') == 'Chiller' %}cool{% elif  states('select.roomcontroller_simulator_state') == 'Heater'%}heat{% endif %}"

    set_hvac_mode: # send the climates current state to esphome.
      - service: select.select_option
        target:
          entity_id: select.roomcontroller_simulator_state
        data:
          option: "{% if hvac_mode == 'off' %}Stop{% elif hvac_mode == 'cool' %}Chiller{% elif hvac_mode == 'heat' %}Heater{% endif %}"

    set_temperature:
      - service: number.set_value
        data:
          value: "{{ temperature }}"
        target:
          entity_id: number.roomcontroller_simulator_setpoint_1
```

I think for your configuration at the select action, you're using a `fan_mode` key where `option` should be used.
For the options, you need to make sure it's a string. All select entities use strings as their options.
From this example, you can substitute Stop, Chiller and Heater with 1, 2 and 3 and it should work as well. Just depends on what value belongs to what mode, but I'm sure that's noted in Daikin documentation.

BradleyFord, 2024-05-09:
Perfect thank you for that, it is now nearly working perfectly.!

My only challenge is "Off" as that is represented by a different entity to the modes, but I will need to create something fancy for that...its a job for me to explore tonight :). Essentially creating a new entity that merges Off/Cool/Dry/Fan

```
    set_hvac_mode: # send the climates current state
      - service: select.select_option
        target:
          entity_id:  select.bacnet_airconmodecommand_020
        data:
          option: "{% if hvac_mode == 'cool' %}1{% elif hvac_mode == 'fan_only' %}3{% elif hvac_mode == 'dry' %}5{% endif %}"

    set_fan_mode: # send the climates current state
      - service: select.select_option
        target:
          entity_id: select.bacnet_airflowratecommand_020
        data:
          option: "{% if fan_mode == 'low' %}1{% elif fan_mode == 'medium' %}2{% elif fan_mode == 'high' %}3{% elif fan_mode == 'auto' %}4{% endif %}"

    set_temperature:
      - service: number.set_value
        data:
          value: "{{ temperature }}"
        target:
          entity_id: number.bacnet_tempadjust_020
```

BradleyFord, 2024-05-15:
Ah I see what you did, I have adapted this to my setup and its working perfectly. Thanks very much!!!

Belown is a sample from one of the ACs; hopefully it is useful for someone.

```
climate:
  - platform: climate_template
    unique_id: l1_ac_parents
    name: L1_AC_parents
    icon_template: mdi:air-conditioner
    hvac_modes:
      - "dry"
      - "off"
      - "cool"
      - "fan_only"
    swing_modes:
      - "off"
      - "vertical"
    fan_modes:
      - "auto"
      - "low"
      - "medium"
      - "high"
    min_temp: 24
    max_temp: 30

    current_temperature_template: "{{ states('sensor.bacnet_roomtemp_001') }}"
    target_temperature_template: "{{ states('number.bacnet_tempadjust_001') }}"
    hvac_mode_template: "{% if states('binary_sensor.bacnet_startstopstatus_001') == 'off' %}off{% elif states('sensor.bacnet_airconmodestatus_001') == '1' %}cool{% elif  states('sensor.bacnet_airconmodestatus_001') == '3'%}fan{% elif  states('sensor.bacnet_airconmodestatus_001') == '5'%}dry{% else %}off{% endif %}"
    fan_mode_template: "{% if states('sensor.bacnet_airflowratestatus_001') == '4' %}auto{% elif  states('sensor.bacnet_airflowratestatus_001') == '1'%}low{% elif  states('sensor.bacnet_airflowratestatus_001') == '3'%}medium{% elif  states('sensor.bacnet_airflowratestatus_001') == '2'%}high{% else %}off{% endif %}"

    set_hvac_mode: # send the climates current state.
      - service: select.select_option
        target:
          entity_id:  select.bacnet_airconmodecommand_001
        data:
          option: "{% if hvac_mode == 'cool' %}1{% elif hvac_mode == 'fan_only' %}3{% elif hvac_mode == 'dry' %}5{% elif hvac_mode == 'off' %}1{% endif %}"
      - service: "{% if hvac_mode == 'off' %}switch.turn_off{% else %}switch.turn_on{% endif %}"
        target:
          entity_id:  switch.bacnet_startstopcommand_001
    set_fan_mode: # send the climates current state.
      - service: select.select_option
        target:
          entity_id: select.bacnet_airflowratecommand_001
        data:
          option: "{% if fan_mode == 'low' %}1{% elif fan_mode == 'medium' %}2{% elif fan_mode == 'high' %}3{% elif fan_mode == 'auto' %}4{% endif %}"
    set_temperature:
      - service: number.set_value
        data:
          value: "{{ temperature }}"
        target:
          entity_id: number.bacnet_tempadjust_001
```

closed 2024-05-15 by BradleyFord