Bepacom-Raalte/bepacom-HA-Addons #66: Add-on will not start.

skeatcherz, 2024-06-15:
I cannot get the add-on to run. Here are the logs I am seeing.  Sorry if it's obvious, I'm still pretty new to HA. Any help would be greatly appreciated. 

-----------------------------------------------------------
 Add-on: Bepacom BACnet/IP Interface
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet/IP devices to be available to Home Assistant through an API.
-----------------------------------------------------------
 Add-on version: 1.4.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (aarch64 / green)
 Home Assistant Core: 2024.6.2
 Home Assistant Supervisor: 2024.06.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[14:57:19] INFO: Running init-interface.
[14:57:19] INFO: Running init-nginx.
[14:57:19] INFO: Allowed addresses for NGINX: 127.0.0.1 192.168.68.68 172.30.232.1 172.30.32.1
/etc/s6-overlay/s6-rc.d/init-interface/run: line 59: ipaddr: unbound variable
s6-rc: warning: unable to start service init-interface: command exited 1
[14:57:20] INFO: Generated ingress configuration successfully!
s6-rc: info: service init-nginx successfully started
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service init-nginx: stopping
s6-rc: info: service init-nginx successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

GravySeal, 2024-06-16:
Hello,

Thank you for your report, it wasn't something you did wrong.
I missed to declare a variable somewhere.
If you check for updates in the add-on store, you can reinstall the current version of the bacnet add-on. The numbers won't update but initialisation shouldn't fail anymore.

thijs86, 2024-07-04:
Hello,
I think i have the same issue, just installed the interface and it won't start.
Can you help me to get it to work,
Thanks a lot.

<details><summary>Log</summary>
<p>

Add-on: Bepacom BACnet/IP Interface
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet/IP devices to be available to Home Assistant through an API.
-----------------------------------------------------------
 Add-on version: 1.4.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (aarch64 / raspberrypi3-64)
 Home Assistant Core: 2024.7.0
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[13:11:10] INFO: Running init-nginx.
[13:11:10] INFO: Running init-interface.
[13:11:10] INFO: Allowed addresses for NGINX: 127.0.0.1 192.168.1.106 172.30.232.1 172.30.32.1
/etc/s6-overlay/s6-rc.d/init-interface/run: line 60: ipaddr: unbound variable
s6-rc: warning: unable to start service init-interface: command exited 1
[13:11:10] INFO: Generated ingress configuration successfully!
s6-rc: info: service init-nginx successfully started
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service init-nginx: stopping
s6-rc: info: service init-nginx successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

</p>
</details>

GravySeal, 2024-07-07:
How about if you install the development version?
Could you perhaps share your add-on configuration with me?

skeatcherz, 2024-07-07:
Sorry I forgot to comment back. I fixed my issue by manually entering my IP address instead of using auto.

thijs86, 2024-07-07:
> Sorry I forgot to comment back. I fixed my issue by manually entering my IP address instead of using auto.

Oh yes, this easy, it works when setting IP adress manually. Sorry, should have figured this one out myself.
Thanks a lot!

thijs86, 2024-07-07:
> How about if you install the development version? Could you perhaps share your add-on configuration with me?

Hello! I had same with the development version. But is working now after setting IP address manually, see comment above.
Thanks!

GravySeal, 2024-07-07:
Auto should now work as well. I found it's because in some recent update, Home Assistant updated systemd to v197+, and thus some ethernet adapter names changed. They should be able to be found now, and if not, handled better to give a warning about this.

closed 2024-08-13 by GravySeal