Bepacom-Raalte/bepacom-HA-Addons #78: Updated to latest beta failing to connect

BradleyFord, 2024-10-26:
HA recommended an update for this, after waiting a few days I hit the trigger. Since then there seems to be no communication from HA.

Any help would be appreciated 👍 

HA Addon log 
```
2024/10/26 08:15:48 [info] 372#372: *621 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:17:18 [info] 372#372: *623 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:18:48 [info] 372#372: *625 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:20:18 [info] 372#372: *627 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:21:48 [info] 372#372: *629 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:23:18 [info] 372#372: *631 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:24:48 [info] 372#372: *633 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:26:18 [info] 372#372: *635 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:27:41 [info] 372#372: *637 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:29:19 [info] 372#372: *639 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:29:51 [info] 372#372: *641 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:30:32 [info] 372#372: *643 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service nginx: stopping
2024/10/26 08:30:58 [notice] 310#310: signal 15 (SIGTERM) received from 27, exiting
2024/10/26 08:30:58 [notice] 372#372: exiting
2024/10/26 08:30:58 [notice] 372#372: exit
2024/10/26 08:30:58 [notice] 310#310: signal 17 (SIGCHLD) received from 372
2024/10/26 08:30:58 [notice] 310#310: worker process 372 exited with code 0
2024/10/26 08:30:58 [notice] 310#310: exit
[08:30:58] INFO: Service NGINX exited with code 0 (by signal 0)
s6-rc: info: service nginx successfully stopped
s6-rc: info: service init-nginx: stopping
s6-rc: info: service interface: stopping
s6-rc: info: service init-nginx successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------

 Add-on: Bepacom BACnet/IP Interface Development Version
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet/IP devices to be available to Home Assistant through an API
-----------------------------------------------------------
 Add-on version: 1.6.0b1
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.0  (amd64 / qemux86-64)
 Home Assistant Core: 2024.10.2
 Home Assistant Supervisor: 2024.10.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[08:31:11] INFO: Running init-interface.
[08:31:11] INFO: Running init-nginx.
[08:31:11] INFO: Allowed addresses for NGINX: 127.0.0.1 192.168.1.222 172.30.32.1 172.30.232.1
[08:31:12] INFO: Generated ingress configuration successfully!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: D-BACS
address: 192.168.1.222/24
objectIdentifier: 420
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
[... 12 lines cut ...]
s6-rc: info: service legacy-services successfully started
[08:31:20]|INFO     |server.py         ->_serve                              : Started server process [305]
[08:31:20]|INFO     |on.py             ->startup                             : Waiting for application startup.
[08:31:20]|INFO     |BACnetIOHandler.py->do_IAmRequest                       : I Am from device,0
[08:31:20]|INFO     |BACnetIOHandler.py->read_multiple_objects               : Reading objects from objectList of device,0...
[08:31:25]|INFO     |on.py             ->startup                             : Application startup complete.
[08:31:25]|INFO     |server.py         ->_log_started_message                : Uvicorn running on http://127.0.0.1:7813 (Press CTRL+C to quit)
[08:31:25] INFO: Running NGINX
[08:31:25]|WARNING  |h11_impl.py       ->handle_events                       : Invalid HTTP request received.
2024/10/26 08:31:25 [notice] 310#310: using the "epoll" event method
2024/10/26 08:31:25 [notice] 310#310: nginx/1.26.2
2024/10/26 08:31:25 [notice] 310#310: OS: Linux 6.6.44-haos
2024/10/26 08:31:25 [notice] 310#310: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/10/26 08:31:25 [notice] 310#310: start worker processes
2024/10/26 08:31:25 [notice] 310#310: start worker process 372
[08:31:28]|ERROR    |BACnetIOHandler.py->create_poll_task                    : Failed to create polling task device,0, all
[08:31:28]|WARNING  |BACnetIOHandler.py->create_poll_task                    : device,0 notification-class,3 failed to read: property: unknown-property
[08:31:28]|INFO     |BACnetIOHandler.py->create_poll_task                    : device,0 notification-class,3 won't get polled.
2024/10/26 08:32:09 [info] 372#372: *1 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/ [... cut ...]
2024/10/26 08:35:12 [info] 372#372: *9 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
2024/10/26 08:36:22 [info] 372#372: *11 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 127.0.0.1, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "127.0.0.1:8099"
```

gioassio, 2024-10-26:
hi, same problem:

[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,61, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,61, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,62, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,62, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,63, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,63, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,64, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,64, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,65, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,65, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,66, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,66, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,67, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,67, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,68, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,68, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,69, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,69, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,70, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,70, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,71, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,71, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,72, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,72, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,73, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,73, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,74, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,74, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,75, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,75, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,76, min-pres-value... Inf value: -inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,76, max-pres-value... Inf value: inf
[15:12:18]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,77, min-pres-value... Inf value: -inf
[... cut ...]
2024/10/26 15:38:42 [info] 362#362: *29 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 172.189.1.12, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "172.189.1.12:8099"
2024/10/26 15:40:33 [info] 362#362: *31 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 172.189.1.12, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "172.189.1.12:8099"

something wrong also in cpu usage:
[image]

GravySeal, 2024-10-26:
If you run regular version v1.5.1, does the communication come back? If so, there might be some packages that don't play nice after the update. Please let me know if the stable version runs well.

BradleyFord, 2024-10-26:
I restored back to 1.5.1 and all seems to be working on again 💪

GravySeal, 2024-10-26:
Great, could you guys test to see if v1.6.0b2 also has this same issue? 
I restored the old versions of all the updated packages in the hope that it's one of those that's causing the issue.

gioassio89, 2024-10-26:
hi, with v1.6.0b2 values can be read but the cpu usage is the same than v1.6.0b1:

[image]

with the v1.5.1 the cpu usage is minimal:
[image]

the device setup is in both case the default:

- deviceID: all
  CoV_lifetime: 600
  CoV_list:
    - all
  quick_poll_rate: 5
  quick_poll_list: []
  slow_poll_rate: 600
  slow_poll_list:
    - all
  resub_on_iam: true
  reread_on_iam: false

Here v1.6.0b2 log:

[23:35:33]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,51, min-pres-value... Inf value: -inf
[23:35:33]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,51, max-pres-value... Inf value: inf
[23:35:33]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,52, min-pres-value... Inf value: -inf
[23:35:33]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,52, max-pres-value... Inf value: inf
[23:35:33]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,53, min-pres-value... Inf value: -inf
[23:35:33]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,53, max-pres-value... Inf value: inf
[23:35:33]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,54, min-pres-value... Inf value: -inf
[23:35:33]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,54, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,55, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,55, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,56, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,56, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,57, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,57, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,58, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,58, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,59, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,59, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,60, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,60, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,61, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,61, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,62, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,62, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,63, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,63, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,64, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,64, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,65, min-pres-value... Inf value: -inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,65, max-pres-value... Inf value: inf
[23:35:34]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,18911, analog-value,66, min-pres-value... Inf value: -inf
[... cut ...]
[23:35:40]|WARNING  |BACnetIOHandler.py->create_poll_task                    : device,18911 trend-log,100907 failed to read: property: unknown-property
[23:35:40]|WARNING  |BACnetIOHandler.py->create_poll_task                    : device,18911 trend-log,100908 failed to read: property: unknown-property
2024/10/26 23:36:32 [info] 362#362: *1 epoll_wait() reported that client prematurely closed connection, so upstream connection is closed too while sending request to upstream, client: 172.189.1.12, server: , request: "GET /apiv1/json HTTP/1.1", upstream: "http://127.0.0.1:7813/apiv1/json", host: "172.189.1.12:8099"

GravySeal, 2025-06-15:
Does the issue still persist if you use this repository and the latest beta in here?

https://github.com/Bepacom-Raalte/bepacom-HA-Addons#FlowControl

BradleyFord, 2025-06-16:
I just forked your repo so I could set this to default; so then I can just use HACS to add it.

It worked perfectly on first start-up, other than copying over the add-on settings everything else just worked out of the box (automations, templates, etc)

Thanks very much!

closed 2025-07-02 by BradleyFord