Bepacom-Raalte/bepacom-HA-Addons #81: No suitable ethernet adapters found

selfstream, 2024-11-12:
I've been running version 1.1.3 for a while now - without any problems. Now I installed HomeAssistant updates and then I could no longer start the add-on. I was also no longer able to view the log files (failed to get 13b6b180_bacnetinterface logs, e.callApiRaw is not a function)

An update to 1.5.1 didn't work either.

So I uninstalled everything and now the add-on starts - but the integration cannot connect and Yabe can't find anything either.

I've already tried the whole thing on a new HomeAssistant installation. Here I have the same behavior. Also with the new beta version

What else can I try?

The log says:
Warning: No suitable ethernet adapters found. You probably won't detect anything now.

Could that be the problem?

```Add-on: Bepacom BACnet/IP Interface
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet/IP devices to be available to Home Assistant through an API.
-----------------------------------------------------------
 Add-on version: 1.5.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.2  (aarch64 / raspberrypi3-64)
 Home Assistant Core: 2024.11.1
 Home Assistant Supervisor: 2024.11.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[17:12:43] INFO: Running init-interface.
[17:12:43] INFO: Running init-nginx.
[17:12:43] INFO: Allowed addresses for NGINX: 127.0.0.1 192.168.178.30 172.30.232.1 172.30.32.1
Using enu1u1u1 as address: 192.168.178.30/24
[17:12:43] INFO: Generated ingress configuration successfully!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.178.30/24
objectIdentifier: 420
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: WARNING
defaultPriority: 15
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
[17:12:44] INFO: Running interface
s6-rc: info: service legacy-services successfully started
Warning: No suitable ethernet adapters found. You probably won't detect anything now.
[17:13:05]|WARNING  |h11_impl.py       ->handle_events                       : Invalid HTTP request received.
[17:13:05] INFO: Running NGINX
2024/11/12 17:13:05 [notice] 299#299: using the "epoll" event method
2024/11/12 17:13:05 [notice] 299#299: nginx/1.26.2
2024/11/12 17:13:05 [notice] 299#299: OS: Linux 6.6.31-haos-raspi
2024/11/12 17:13:05 [notice] 299#299: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/11/12 17:13:05 [notice] 299#299: start worker processes
2024/11/12 17:13:05 [notice] 299#299: start worker process 377
```

GravySeal, 2024-11-12:
Hello,
Could you perhaps share the configuration you put in the add-on?
Have you tried manually adding your IP address into the add-on?
In the web UI, there's also an option to download logs. Could you please share these with me too?

selfstream, 2024-11-12:
Thank you. Yes, i already tried it.  But your right - then there is another error:

```
-----------------------------------------------------------
 Add-on: Bepacom BACnet/IP Interface
 Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet/IP devices to be available to Home Assistant through an API.
-----------------------------------------------------------
 Add-on version: 1.5.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.2  (aarch64 / raspberrypi3-64)
 Home Assistant Core: 2024.11.1
 Home Assistant Supervisor: 2024.11.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[21:36:56] INFO: Running init-nginx.
[21:36:56] INFO: Running init-interface.
[21:36:56] INFO: Allowed addresses for NGINX: 127.0.0.1 192.168.178.30 172.30.232.1 172.30.32.1
[21:36:56] INFO: Generated ingress configuration successfully!
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.178.30/24
objectIdentifier: 420
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
foreignBBMD: -
foreignTTL: 255
maxSegmentsAccepted: 64
loglevel: WARNING
defaultPriority: 15
s6-rc: info: service init-interface successfully started
s6-rc: info: service interface: starting
s6-rc: info: service interface successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[21:36:58] INFO: Running interface
[21:37:13]|ERROR    |main.py           ->exception_handler                   : An uncaught error occurred: no broadcast
NoneType: None
[21:37:13]|ERROR    |main.py           ->exception_handler                   : An uncaught error occurred: no broadcast
NoneType: None
[21:37:18]|WARNING  |h11_impl.py       ->handle_events                       : Invalid HTTP request received.
[21:37:18] INFO: Running NGINX
2024/11/12 21:37:18 [notice] 311#311: using the "epoll" event method
2024/11/12 21:37:18 [notice] 311#311: nginx/1.26.2
2024/11/12 21:37:18 [notice] 311#311: OS: Linux 6.6.31-haos-raspi
2024/11/12 21:37:18 [notice] 311#311: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/11/12 21:37:18 [notice] 311#311: start worker processes
2024/11/12 21:37:18 [notice] 311#311: start worker process 388
```

My config:
```
objectName: EcoPanel
address: 192.168.178.30
objectIdentifier: 420
defaultPriority: 15
devices_setup:
  - deviceID: all
    CoV_lifetime: 600
    CoV_list:
      - all
    quick_poll_rate: 5
    quick_poll_list: []
    slow_poll_rate: 600
    slow_poll_list:
      - all
    resub_on_iam: true
    reread_on_iam: false
entity_list: []
api_accessible: false
loglevel: WARNING
segmentation: segmentedBoth
```

and the log:
```
[21:37:12]|DEBUG    |BACnetIOHandler.py->__init__                            : Application initialised
[21:37:12]|DEBUG    |connectionpool.py ->_new_conn                           : Starting new HTTP connection (1): supervisor:80
[21:37:13]|DEBUG    |connectionpool.py ->_make_request                       : http://supervisor:80 "GET /core/api/services HTTP/11" 200 67075
[21:37:13]|ERROR    |main.py           ->exception_handler                   : An uncaught error occurred: no broadcast
NoneType: None
[21:37:13]|ERROR    |main.py           ->exception_handler                   : An uncaught error occurred: no broadcast
NoneType: None
[21:37:18]|WARNING  |h11_impl.py       ->handle_events                       : Invalid HTTP request received.
[21:49:20]|DEBUG    |webAPI.py         ->websocket_endpoint                  : Accepted websocket: ws://192.168.178.30:8123/api/hassio_ingress/5lPaze2GqLhe2Bvo9kR1OHPajf8SKGRDbBWi9AtpFHQ/ws
[21:49:20]|DEBUG    |webAPI.py         ->websocket_writer                    : Passed send_json test
[21:49:22]|DEBUG    |webAPI.py         ->websocket_endpoint                  : Data received: {'type': 'websocket.disconnect', 'code': 1000}
[21:49:22]|INFO     |webAPI.py         ->websocket_endpoint                  : Disconnected websocket: 
[21:49:22]|DEBUG    |webAPI.py         ->websocket_writer                    : Websocket writer cancelled: 
```

selfstream, 2024-11-12:
Oh my god. Thank you so much! I spent so many hours because of my mistake.

GravySeal, 2024-11-12:
Glad it's solved!