Bepacom-Raalte/bepacom-HA-Addons #82: ErrorRejectAbortNack

selfstream, 2024-11-13:
Sorry, i have the next Problem.... Do you know what i have to do?

```
[13:11:59]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,2640, binary-value,20043: resources: no-space-to-add-list-element
[13:11:59]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,2640, binary-value,20044: resources: no-space-to-add-list-element
[13:11:59]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,2640, binary-value,20045: resources: no-space-to-add-list-element
[13:11:59]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,2640, binary-value,20046: resources: no-space-to-add-list-element
[13:11:59]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,2640, binary-value,20055: resources: no-space-to-add-list-element
[13:11:59]|ERROR    |BACnetIOHandler.py->subscription_task                   : ErrorRejectAbortNack: device,2640, binary-value,20056: resources: no-space-to-add-list-element
```

selfstream, 2024-11-13:
Okay, thanks. That was the next step!
Now I have the next problem. But I'm not sure if it's with the addon or with the interface.  

After restarting HomeAssistant, all entities are available. But after polling, none are available anymore and i have to restart again.

same problem: when i change an option in a select list all entities are not available and i have to restart HA (at the end of the log)

Do you have any idea?

```
objectName: EcoPanel
address: 192.168.178.30/24
objectIdentifier: 420
defaultPriority: 15
devices_setup:
  - deviceID: device:2640
    CoV_lifetime: 900
    CoV_list:
      - multiStateValue:30368
      - multiStateValue:40003
      - multiStateValue:40284
    quick_poll_rate: 20
    quick_poll_list: []
    slow_poll_rate: 500
    slow_poll_list:
      - all
    resub_on_iam: true
    reread_on_iam: false
entity_list: []
api_accessible: false
loglevel: WARNING
segmentation: segmentedBoth
```

```

[34m-----------------------------------------------------------[0m
[34m Add-on: Bepacom BACnet/IP Interface[0m
[34m Bepacom BACnet/IP interface for the Bepacom EcoPanel. Allows BACnet/IP devices to be available to Home Assistant through an API.[0m
[34m-----------------------------------------------------------[0m
[34m Add-on version: 1.5.1[0m
[32m You are running the latest version of this add-on.[0m
[34m System: Home Assistant OS 13.2  (aarch64 / raspberrypi3-64)[0m
[34m Home Assistant Core: 2024.11.1[0m
[34m Home Assistant Supervisor: 2024.11.2[0m
[34m-----------------------------------------------------------[0m
[34m Please, share the above information when looking for help[0m
[34m or support in, e.g., GitHub, forums or the Discord chat.[0m
[34m-----------------------------------------------------------[0m
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-interface: starting
[23:14:22] INFO: [32mRunning init-nginx.[0m
[23:14:22] INFO: [32mRunning init-interface.[0m
[23:14:22] INFO: [32mAllowed addresses for NGINX: 127.0.0.1 192.168.178.30 172.30.232.1 172.30.32.1[0m
[23:14:23] INFO: [32mGenerated ingress configuration successfully![0m
s6-rc: info: service init-nginx successfully started
[BACpypes]
objectName: EcoPanel
address: 192.168.178.30/24
objectIdentifier: 420
maxApduLengthAccepted: 1476
segmentation: segmentedBoth
vendorIdentifier: 15
[... 149 lines cut ...]
[23:23:28]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40020, present-value... NaN value: nan
[23:23:28]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40023, present-value... NaN value: nan
[23:23:28]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40024, present-value... NaN value: nan
[23:23:28]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40025, present-value... NaN value: nan
[23:23:28]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40135, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40150, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40151, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40152, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40153, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40155, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40286, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40287, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40288, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40289, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40290, present-value... NaN value: nan
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40291, present-value... NaN value: nan
[... cut ...]
[23:23:29]|WARNING  |BACnetIOHandler.py->dict_updater                        : Ignoring property: device,2640, analog-value,40329, present-value... NaN value: nan
[23:25:46]|ERROR    |webAPI.py         ->write_property                      : device,2640, multi-state-value,30368, present-value, 3, None
[23:25:46]|ERROR    |webAPI.py         ->write_property                      : device,2640, multi-state-value,30368, present-value, 2, None
2024/11/13 23:26:18 [info] 388#388: *10 client 192.168.178.30 closed keepalive connection
2024/11/13 23:26:18 [info] 388#388: *11 client 192.168.178.30 closed keepalive connection
```

GravySeal, 2024-11-14:
Could you please share some Home Assistant logs, as well as download the logs through the web UI? I think it might be going wrong in the integration.

selfstream, 2024-11-14:
Yes, i think the problem ist with the integration. With the web UI for example i can send presentValue. 

I found this log in HomeAssistant:

```
Logger: homeassistant.components.number
Quelle: helpers/entity_platform.py:599
Integration: Zahl (Dokumentation, Probleme)
Erstmals aufgetreten: 20:24:12 (24 Vorkommnisse)
Zuletzt protokolliert: 20:24:13

Error adding entity number.corrigoventilation_extrasafpressure for domain number with platform bacnet_interface
Error adding entity number.corrigoventilation_extraeafpressure for domain number with platform bacnet_interface
Error adding entity number.corrigoventilation_filterguard1ai for domain number with platform bacnet_interface
Error adding entity number.corrigoventilation_filterguard2ai for domain number with platform bacnet_interface
Error adding entity number.corrigoventilation_efficiencytemp for domain number with platform bacnet_interface
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 599, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 918, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1368, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1007, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1069, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 359, in state
    return self.value
           ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 398, in value
    if (native_value := self.native_value) is None:
                        ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bacnet_interface/number.py", line 343, in native_value
    value = float(
            ^^^^^^
TypeError: float() argument must be a string or a real number, not 'NoneType'
```
and this

```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: homeassistant
Quelle: custom_components/bacnet_interface/number.py:343
Integration: Bepacom BACnet/IP Interface (Dokumentation, Probleme)
Erstmals aufgetreten: 20:25:13 (10 Vorkommnisse)
Zuletzt protokolliert: 20:34:58

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 266, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 481, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 176, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 561, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1007, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1069, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 359, in state
    return self.value
           ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 398, in value
    if (native_value := self.native_value) is None:
                        ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bacnet_interface/number.py", line 343, in native_value
    value = float(
            ^^^^^^
TypeError: float() argument must be a string or a real number, not 'NoneType'
``` 

The bacnet addon log looks good for me.
[bacnet_addon_logs.txt](https://github.com/user-attachments/files/17756850/bacnet_addon_logs.txt)
[home-assistant_2024-11-14T19-35-35.689Z.log](https://github.com/user-attachments/files/17756876/home-assistant_2024-11-14T19-35-35.689Z.log)

GravySeal, 2024-11-14:
I think I kind of get an idea what's happening, could you perhaps share another file with me?
If you go to the web UI, to Swagger API Docs, and click on /apiv1/json, you should be able to get the data the add-on sends to Home Assistant. Could you press "Try it out" and then execute and download that file? It should tell me what I need to know.

Could you perhaps also try to remove the slow poll list, the same was as quick poll list. Does the problem come back if you remove polling altogether?

selfstream, 2024-11-14:
The JSON-File:
[response_1731622951861.json](https://github.com/user-attachments/files/17759053/response_1731622951861.json)

I have played around:
When I remove the slow poll list with "slow_poll_list: []". The problem still appears after a few minutes. But I think the time is never the same. After 3-10 minutes the entities are no longer available.

But when i also remove the CoV_list, it seems that there are no problems (until I try to change a value).

GravySeal, 2024-11-15:
I think I found an issue, if you look in the add-on logs you can see the add-on ignores a couple of analog value present values. This is because they come in as NaN. Could you check through Yabe or another BACnet program what these values are? For example for analog value 40010?

selfstream, 2024-11-15:
I can't start working with Yabe until this evening. But on the main page of the add-on I can see that all of these analog values ​​that are ignored don't have a present value. Is there any way I can exclude these from polling?

[image]

selfstream, 2024-11-21:
[image]

GravySeal, 2024-11-23:
Hmm, it's truly shown as NaN. Let me see if I can handle these values in another way.
As for excluding these objects, there isn't a blacklist yet. 
The only way now is to add all objects you want polled to the configuration, excluding the objects that don't work.