Bepacom-Raalte/bepacom-HA-Addons #83: Some device values wrong in Samsung DMS setup

DevRalf, 2025-01-11:
Just getting started with BACnet and Home Assistant to monitor the performance of my Samsung VRF HVAC system (using a Samsung DMS controller). Got the Bepacom add-on and interface all up and running and its great!
I can see a lot of the values I am looking for (mostly various temperature values) and use them for some nice graphs on my HA dashboards.

The problem is that _some_ values are '-1000' instead of the actual temperature value. This happens on all my devices of that class (indoor units). 

This happens for example for the following analog data objects in my system:
- ObjectID: 22 (analog value) "AC_DisCoolTemp_Set", actual value (confirmed directly with Samsung DMS): 59℉, reported by Bepacom: '-1000'
- ObjectID: 23 (analog value) "AC_DisHeatTemp_Set", actual value: 100℉, reported as '-1000'
- ObjectID: 24 (analog input) "AC_DisCurrentTemp", actual value: 78℉, reported as '-1000'

I checked these values directly in Home Assistant's Entity information, inside the Bepacom Interface UI in HA and also using the Web API - getting '-1000' in all three places.

I checked out the Bepacom logs but didn't see anything 'suspicious'.

I am using v 1.5.1 of the add-on on Home Assistant 2025.1.2

Not sure what else to try? Any help would be greatly appreciated.

[image]
[bacnet_addon_logs (1).txt](https://github.com/user-attachments/files/18387665/bacnet_addon_logs.1.txt)

GravySeal, 2025-01-12:
Hey @DevRalf,

That's strange, I don't see anything strange happening in the add-on logs.how about if you scan Samsung device with something like Yabe?
Does the present value also report -1000 there or does it show correctly?

DevRalf, 2025-01-12:
Hey @GravySeal,
I just installed BDT (BACnet Discovery Tool) on an old Windows box to get another tool to look at the same data and it looks like this might be a Samsung issue: BDT reports the same value ('-1000') instead of the actual temperature values.
I guess there is not much else from your end we could try?, right? Let me know if you want me to just close this ... and thanks for your help!

GravySeal, 2025-01-13:
If other BACnet tools report the same values, I guess it's a Samsung issue then. Not much more we can do about it other than report it to Samsung. 
As for this issue, feel free to close it!

closed 2025-01-13 by DevRalf