Bepacom-Raalte/bepacom-HA-Addons #89: All Bacnet entities became unavailable after updating to 1.6.0b3 and 1.6.0b4.

CV-Private, 2025-03-26:
The Add-On 1.6.0b2 was working fine. Then, last week, I updated to 1.6.0b3, which broke all Bacnet entities in HA. A day or two later, the update to 1.6.0b4 was released, but it didn't change anything. None of the Bacnet entities are available anymore.

- I didn't change anything else in the Add-On configuration. 
- I restarted the Add-On (many times)
- I restarted HA (many times) 
- I disconnected and reconnected the Bacnet device, 
- I rebuilt the Add-On.
- I checked with Yabe that the Bacnet device is doing fine.

Nothing helped yet - the entities remain unavailable:

[image]

The Bacnet IP-Interface actually appears to work, because I do see the Bacnet device and objects, and the subscriptions are confirmed, but the connection to the entities seems to be completely broken.

[image]

Please advise how to fix it, or what else you need from me.

Thanks and best regards,
Christian

GravySeal, 2025-03-26:
Hey Christian, could you please post Home Assistant entity logs? This would give me a direction in where to look

CV-Private, 2025-03-26:
I don't see any Log entries from Bacnet entities, unfortunately:

[image]

Also not here, no matter what Bacnet entity I'm selecting:

[image]

CV-Private, 2025-03-26:
The note in every Bacnet entity says: "This entity is no longer provided by the bacnet_interface integration. If the entity is no longer used, delete it in the settings."

[image]

GravySeal, 2025-03-26:
In Home Assistant, if you go to Settings->System->Logs, you should see all integration warnings and errors that have happened. If you could share some downloaded logs from there, that'd be great.

GravySeal, 2025-03-26:
Thanks for reporting 😜 
In the add-on, could you please go to swagger API docs->/apiv1/json->Try it out->execute, download this file and share it with me?
Then I should be able to figure out what's going wrong.

CV-Private, 2025-03-26:
Here it is:

[response_1743028375930.json](https://github.com/user-attachments/files/19476246/response_1743028375930.json)

GravySeal, 2025-03-27:
So I think I know what's going on, but to be sure, could you restart the add-on, let it run for about 10-20 minutes and then download the logs through the web UI?

CV-Private, 2025-03-28:
Here's the UI log, 20 min after HA restart:

[bacnet_addon_logs.txt](https://github.com/user-attachments/files/19509823/bacnet_addon_logs.txt)

Also the Swagger log, in case that you need it:

[response_1743182611000.json](https://github.com/user-attachments/files/19509829/response_1743182611000.json)

GravySeal, 2025-03-28:
Could you add this repository to your add-on store?
https://github.com/Bepacom-Raalte/bepacom-HA-Addons#MissingObjectIdentifier

Then download the development add-on and try again please.

CV-Private, 2025-03-28:
I hope I did it right. I don't see a difference in HA yet. Entities are still unavailable.

Here are the logs i gathered:

[bacnet_addon_logs(1).txt](https://github.com/user-attachments/files/19511516/bacnet_addon_logs.1.txt)

[response_1743191143072.json](https://github.com/user-attachments/files/19511518/response_1743191143072.json)

GravySeal, 2025-03-28:
If you check for updates in the add-on store, there will be a 1.6.0b5 for the new branch, could you check it out?

CV-Private, 2025-03-29:
I tried the 1.6.0b5 yesterday. At that time, the Bacnet entities were still unavailable, but I could at least see the bacnet device in the web UI of add-on 1.6.0b5.

Today, just half an hour ago, I installed the latest HA Core and HA OS updates 25.03. Now, after restart, there's no more device in the web UI at all, and the Log shows Websocket errors:

[bacnet_addon_logs(2).txt](https://github.com/user-attachments/files/19522107/bacnet_addon_logs.2.txt)

[image]

[image]

GravySeal, 2025-03-29:
Could it be that another version of the add-on tried to automatically start on reboot? Could you check and restart the add-on and try again?

CV-Private, 2025-03-29:
I have 1.6.0b4 and 1.6.0b5 installed, and I made sure that only one of them is running at a time. I just checked the older 1.6.0b4 Add-On, and that does also not find a Bacnet device (it did find it before the latest HA updates).

GravySeal, 2025-03-29:
Could you try to rebuild the add-on and check the configuration to be sure?

CV-Private, 2025-03-29:
The device is back! I had checked with Yabe and didn't see it either. Then pulled the LAN from the Bacnet device, reconnected, and now its back and 1.6.0b5 discovered it. Sorry for the confusion - not sure what was going on...

So, would you expect that I would see the entities if 1.6.0b4 and 1.6.0b5 are both installed, but only 1.6.0b5 is running, whereas the entities are tied to 1.6.0b4? 

Or should I remove the old one? I would be real pity if I'd have to recreate all entity settings, because it took very long time. I'm hoping that the old Add-On can be fixed so that the existing entities will remain...

CV-Private, 2025-03-29:
Well, somethings not right, though... The Add-On 1.6.0b5 only finds the device but no objects. There's an Abnormal_Closure in the Log:

[bacnet_addon_logs(7).txt](https://github.com/user-attachments/files/19522385/bacnet_addon_logs.7.txt)

[image]

GravySeal, 2025-03-29:
You should be able to keep using the old entities, so I am not sure what's going on there. 
Did you check for updates in the add-on store and rebuilt the add-on? If v1.6.0b5 works after that, I recommend just keeping that installed for a while.

CV-Private, 2025-03-29:
Did that, but no updates were found. None of the two versions are currently "working", because the entities are unavailable. Can I go back to 1.6.0b2? I think that was the one that ran fine.

GravySeal, 2025-03-29:
If you have 1.6.0b2 backed up, you can go back. But no new update will show up when you check for updates, it just helps Home Assistant definitely get the last version from GitHub when you rebuild the add-on. The rebuild option is next to "Open Web-UI" and Delete on the add-on page. Please try that.

CV-Private, 2025-03-29:
I had 1.6.0b1 backed up. Restored it, restarted HA, can't open the UI anymore. :-(

[image]

Rebuild was denied - I could only update to 1.6.0b4. I did that, restarted, and the UI still says 503 Service Unavailable.

It's getting worse, the more I try..

CV-Private, 2025-03-29:
I restored 1.6.0b1 from a backup. That's the last one I backed up. I was letting it run for a while, but discovery stalls at device 1. I have now updated to 1.6.0b4 again, rebuilt, restarted HA, and waiting now. Looks again that the discory stalls and the entities are unavailable, but I'll give it more time.

Here's the Log so far: 

[bacnet_addon_logs(12).txt](https://github.com/user-attachments/files/19522702/bacnet_addon_logs.12.txt)

GravySeal, 2025-03-29:
Could you try v1.6.0b5 on the custom branch? V1.6.0b4 hasn't changed yet

CV-Private, 2025-03-29:
v1.6.0b5 appears to stall early in the discovery. It's not finding anything beyond device 1:

[bacnet_addon_logs(15).txt](https://github.com/user-attachments/files/19522745/bacnet_addon_logs.15.txt)

v1.6.0b4 can at least discover objects and confirm CoV subscriptions.

CV-Private, 2025-03-29:
v1.6.0b5 even appear to crash.. the web UI is no longer being displayed (gray window instead):

[image]

GravySeal, 2025-03-29:
It does take a while to get the entire device read, since it's a quite slow connection for a large device. If you have some patience, does anything eventually appear?

CV-Private, 2025-03-29:
With  1.6.0b5, I was waiting much longer than I usually have to wait for the discovery (typ. 15-20 min), and when I checked again, the UI was blank (the Add-On was still showing some CPU activity, though).

What's worse, though, is, that the Add-On 1.6.0b5 appears to impact the Bacnet device! When the UI got blank, the GTW-21 was disappeared from the network and I couldn't even see it in Yabe anymore, until I unplugged and reconnected the LAN cable of the GTW-21. I NEVER saw this happening before I installed 1.6.0b5.

Since I can't do anything useful with 1.6.0b5 for the time being, I have now 1.6.0b4 running again, and there's another reproducible effect: the device discovery stops after each startup at Analog Input 154, and this is also visible in the log:

[bacnet_addon_logs(20).txt](https://github.com/user-attachments/files/19522946/bacnet_addon_logs.20.txt)

At least, the GTW-21 doesn't get silenced by 1.6.0b4, and I can still read all objects with Yabe without issues:

[image]

GravySeal, 2025-03-30:
I spent a while rewriting parts of the add-on, if you want you can try it. No guarantees it'll work, but any logs you could share with me will help.
You can download it from this repository at the dev version.
https://github.com/Bepacom-Raalte/bepacom-HA-Addons#Rewrite

CV-Private, 2025-03-30:
I just installed it (including rebuilt). Is it correct that it still says 1.6.0b5?

The web UI doesn't find the Bacnet device (the GTW-21 is visible for Yabe, though):

[image]

CPU activity goes to 50% at startup, but very soon to 0.1%:

[image]

Web-UI log:

[bacnet_addon_logs(3).txt](https://github.com/user-attachments/files/19527592/bacnet_addon_logs.3.txt)

Add-On Protocol:

[a9c09ad8_bacnetinterface_dev_2025-03-30T21-27-38.008Z.log](https://github.com/user-attachments/files/19527596/a9c09ad8_bacnetinterface_dev_2025-03-30T21-27-38.008Z.log)

GravySeal, 2025-04-04:
Could you try the regular branch?

CV-Private, 2025-04-05:
With the 1.6.0b5 version of the regular branch, Device 1 is being discovered, but not yet any objects. Something's still stuck.

Here's the GUI log:

[bacnet_addon_logs(9).txt](https://github.com/user-attachments/files/19614844/bacnet_addon_logs.9.txt)

And the Add-On protocol:

[a9c09ad8_bacnetinterface_dev_2025-04-05T10-01-16.567Z.log](https://github.com/user-attachments/files/19614842/a9c09ad8_bacnetinterface_dev_2025-04-05T10-01-16.567Z.log)

CV-Private, 2025-04-05:
I'm not sure where I can see that host name... I'm hoping that this is now the right add-on. It discovers objects up to AI2138 and stops then. No entities are vailable yet.

GUI-log:

[bacnet_addon_logs(12).txt](https://github.com/user-attachments/files/19615078/bacnet_addon_logs.12.txt)

Add-On protocol:

[a9c09ad8_bacnetinterface_dev_2025-04-05T11-18-21.823Z.log](https://github.com/user-attachments/files/19615079/a9c09ad8_bacnetinterface_dev_2025-04-05T11-18-21.823Z.log)

Bepacom-Raalte, 2025-04-05:
You can see which version you're running by the URL or on the add-on page:

[image]

Easiest thing to do is remove of the bacnet add-ons and just re-add the regular repository and install that version.

CV-Private, 2025-04-05:
I deinstalled all Add-On versions and rmoved all repositories from my HA. Then I installed the standard repository from the Github again. After installation, I restarted HA. Here's the screenshot of the Add-On:

[image]

Unfortunately, no device is being discovered by the Add-On (Yabe does find it).

GUI log:

[bacnet_addon_logs(14).txt](https://github.com/user-attachments/files/19615375/bacnet_addon_logs.14.txt)

Add-on protocol:

[97683af0_bacnetinterface_dev_2025-04-05T12-40-05.167Z.log](https://github.com/user-attachments/files/19615376/97683af0_bacnetinterface_dev_2025-04-05T12-40-05.167Z.log)

GravySeal, 2025-04-05:
I see no I-Am response to the who-is request of the add-on during startup. Did you have your device connected to the network before starting the add-on? If not, please restart the add-on and check again for a response. If you could, please also take a peek with Wireshark filtered to BACnet

CV-Private, 2025-04-05:
Yes, the Bacnet device was connected when I started the Add-On. 
To be sure, I did a new test. Here are the steps that I did:

1) I stopped the Add-on.

2) I started Yabe and started the device discovery, while recording with Wireshark. I stopped the discovery after some seconds, as Yabe was finding the objects, and there are so many. Here's the Wireshark log:

[Yabe.zip](https://github.com/user-attachments/files/19615750/Yabe.zip)

3) I started the Add-on. It does find device 1 and objects now.  However, unlike with Yabe, I don't see a lot of Bacnet messages in the Wireshark log:

[Bepacom.zip](https://github.com/user-attachments/files/19615813/Bepacom.zip)

The GUI log also looks odd (so many no response warnings...):

[bacnet_addon_logs(17).txt](https://github.com/user-attachments/files/19615817/bacnet_addon_logs.17.txt)

Add-on protocol:

[97683af0_bacnetinterface_dev_2025-04-05T14-11-06.944Z.log](https://github.com/user-attachments/files/19615819/97683af0_bacnetinterface_dev_2025-04-05T14-11-06.944Z.log)

4) I started Yabe again. Now it cannot discover the GTW-21 anymore. This is very strange - it's like the Add-On would do something that makes the GTW-21 becoming unresponsive. 
I have never seen that happening before I updated the Add-on to 1.6.0b5. The GTW-21 was always discoverable. 
Now it appears as if something in the Add-On would be silencing with device. I can only recover it by unplugging and re-connecting the LAN cable from the Bacnet device.

CV-Private, 2025-04-05:
Oh, I should have mentioned that Yabe and HA are not running on the same machine. They're in the same subnet, and I'm not aware that anything would restrict the network access of HA.
However, nothing in the network setup has changed since I updated the Add-On from v1.6.0b2 to v1.6.0.b3, therefore I'm suspecting that the issue is in the Add-On, rather than anywhere else.

Bepacom-Raalte, 2025-04-05:
The new version does a maximum of 20 requests concurrently, which speeds up reading massively.
Your device might not be able to handle it, so I made an option in this repository to limit the amount of requests at the same time:

https://github.com/Bepacom-Raalte/bepacom-HA-Addons#FlowControl

Please add this repository and set the following settings:

Segmentation to noSegmentation
Concurrent requests to 1

Then make sure your device is ready to accept requests and start the add-on and see what happens.

CV-Private, 2025-04-05:
[image]

[image]

It's discovering Device 1 but no objects.

GUI-log:

[bacnet_addon_logs(29).txt](https://github.com/user-attachments/files/19618683/bacnet_addon_logs.29.txt)

Add-on protocol:

[e1372e08_bacnetinterface_dev_2025-04-05T22-23-32.490Z.log](https://github.com/user-attachments/files/19618684/e1372e08_bacnetinterface_dev_2025-04-05T22-23-32.490Z.log)

GravySeal, 2025-04-06:
Great to hear, if everything seems to work like this you could turn the setting up to 2 or 3. This would speed up the start of the add-on by quite a lot if your device can handle it.
Integration works after every object has been read?

CV-Private, 2025-04-06:
I left the device on during the night. Discovery stopped after AI167 and now the GTW-21 is unresponsive again (I don't see it in Yabe).

[image]

When I look at the GUI log, I see in the first part, that the _properties_read_ was completed. IV7498 is the last object of the GTW-21 device, as far as I remember.

[00:31:37]|DEBUG    |BACnetIOHandler.py->properties_read                     : Read: device,1 positive-integer-value,7498

The GTW-21 appears to become unresponsive when the Add-On starts doing the _[00:36:54]|WARNING  |BACnetIOHandler.py->read_property_safely_ activities.

GUI log:

[bacnet_addon_logs(33).txt](https://github.com/user-attachments/files/19620151/bacnet_addon_logs.33.txt)

Add-on protocol:

[e1372e08_bacnetinterface_dev_2025-04-06T07-20-08.923Z.log](https://github.com/user-attachments/files/19620159/e1372e08_bacnetinterface_dev_2025-04-06T07-20-08.923Z.log)

CV-Private, 2025-04-06:
Not sure if it helps, but I captured a Wireshark log of the GTW-21 discovery made by Yabe. The initial discovery takes 1:05 minutes until completion.

[Yabe_complete.zip](https://github.com/user-attachments/files/19620571/Yabe_complete.zip)

I'm assuming that this is like the _properties_read_ of the Add-On?

When I then mark the entire list of all discovered objects in Yabe, a comprehensive scan begins. It takes about 18 minutes to complete.

I can see that the GTW-21 doesn't like the _ReadPropertyMultiple_ requests from Yabe, but the _ReadProperty_ scan works fine, runs until the final object IV-7498 and the GTW-21 doesn't become unresponsive.

Here's the Wireshark log of the full scan:

[Yabe_full-scan.zip](https://github.com/user-attachments/files/19620688/Yabe_full-scan.zip)

GravySeal, 2025-04-06:
Could you provide me with a wireshark file just like the Yabe full-scan.zip, but then of the add-on and the device communicating? I am curious about the timestamps and want to compare. 
Once Yabe is going automatically, it does send requests quite rapidly.

CV-Private, 2025-04-06:
That's unfortunately challenging for me..  HA runs on a Proxmox VE and i cannot record Wireshark logs on that machine (or at least, I don't know how.

CV-Private, 2025-04-06:
This is my configuration. Unfortunately, I lost all of my previous settings in one of the experiments. Now I need to start all over again.

[image]

GravySeal, 2025-04-06:
Is it possible to get some kind of system log from your device through it's webpage?

CV-Private, 2025-04-06:
Will try, but need to search a bit how to do that.

Currently, I see another problem: I can't open the GUI anymore after some time. The Add-On page is still up and shows activity, but the GUI screen is plain gray.

Here's the Add-On protocol:

[e1372e08_bacnetinterface_dev_2025-04-06T18-44-09.136Z.log](https://github.com/user-attachments/files/19622885/e1372e08_bacnetinterface_dev_2025-04-06T18-44-09.136Z.log)

GravySeal, 2025-04-06:
When the web UI remains gray, it might mean that the add-on is still busy reading. You could turn the logging to debug to keep tabs on whether anything is happening or not. Might have to leave it running for a while before anything happens. Might also be that you need some patience before anything shows on the web UI.

CV-Private, 2025-04-06:
I'm trying to run tcpdump on the Proxmox VM while the Add-on is working. Not sure yet how that will go... will see later...

GravySeal, 2025-04-06:
In this capture I can see BACnet traffic happening as it should be. Your device is responding to every read request, so it does seem to be going on. For information, logging should be set to debug to see if anything is happening at all. A request one by one is very slow, so of course it'll take some time. Maybe your device could handle 2 concurrent requests, or maybe 3. 
New GUI is being worked on, it just takes some time unfortunately.

CV-Private, 2025-04-06:
Good to hear that the traffic is fine. The GUI is still gray. Earlier today, I power-cycled the heating, which also restarts the GTW-21. Maybe that has unclogged something. When the GUI comes back and if all objects are there, I'll try to increase the concurrent requests.

GravySeal, 2025-04-06:
There should be no reason for the web ui to remain gray, unless the CPU is at high usage. Do you have high CPU usage?

CV-Private, 2025-04-06:
The GUI remained gray for the last 3 hours. CPU activity is around 50%:

[image]

Normally, when I start the Add-On, I see CPU activity close to 50% at first, but then it drops to 1-4%. However, when the GUI is gray, it remains around 50%, but still with some slight variation.

I just stopped and restarted the Add-On. That has brought the GUI back.

Interestingly, it shows objects up to analogValue:7235 right after ther restart. I guess that's from a previously stored database? That was also the last objects before the GUI was frozen...

[image]

CV-Private, 2025-04-07:
I had stopped the Add-On yesterday after hours of frozen GUI. I had just started it again, and noticed something new:

The entities were back for some time after the restart!

Initially, the CPU load of the Add-on was around 50%, then dropped below 10%.

Meanwhile, however, maybe half an hour after the restart, the GUI is plain gray again, and all Bacnet entities became unavailable. The CPU usage went up to 50%.

I'll leave it running now and will take a look again tomorrow.

Bepacom-Raalte, 2025-04-08:
I found something in the code that could possibly cause this issue. I bumped the version number to 1.6.0b7, could you try this version to see if this solves the issue?

CV-Private, 2025-04-08:
I updated to 1.6.0b7 (steps: stop, update, rebuild, start). It's looking really good so far. The GUI is there, the entities are available, and all objects have been discovered.

CV-Private, 2025-04-08:
I did notice that the Add-On has subribed to the first 19 objects of the GTW-21, even though I have currently no COV entries in the configuration:

[image]

I was trying to remove the subscriptions, but that doesn't have an effect.

I'm going to configure the Add-On as per my needs and will restart it.

CV-Private, 2025-04-08:
By the way, the Reset to Default button has no effect. It turns green when I click on it, but the settings don't change (also not after restarting the Add-On):

[image]

CV-Private, 2025-04-08:
Update: after power-cycling the heating and restarting the Add-On with my configuration, it's currently working. The CoV subscriptions are confirmed, and the polling works.

I'm wondering, though, why there are many "out-of-resources" errors in the log bacnet_addon_logs(53).txt, such as:

[14:30:07]|WARNING  |BACnetIOHandler.py->read_property_safely                : Error during read: device,1 analog-input,2135 **out-of-resources**

Where is the resource problem? In the GTW-21 device? Or in the Add-On? It seems that the _read_property_safely_ activity clogs up something, and then only a restart helps.

GravySeal, 2025-04-08:
The out of resources error is a response from your device, likely meaning it's overloaded. I think you should dial the concurrency down to 2 or 1, as otherwise it's too much for your device to handle. Once you got it running without such errors, could you maybe post a log of it running?

[3 more replies]